BiancoRoyal/node-red-contrib-modbus #239: Unable to read more than 10 - 12 addresses with Modbus Read / Modbus Flex Read when using RTU

tproulund, 2021-03-15:
Hallo

I recently changed a demo I am building from using TCP to RTU, because I will be adding more device to the network later, which only supports RTU.

Previously I was reading 62 addresses in 1 go with the Modbus read using TCP, but now that i switched to RTU, I am unable to read more than 10-12 addresses. If I increase the number above 12, then I can see on the slave device via a diode, that a communication msg is recieve but i dont get anything back on node red, no errors or anything, I just never get a msg output from the modbus read

I can also confirm same issue if I use the flex reader node

I tried installing the modbus-serial package, and with that one, I have no problem reading the 62 addresses in one Go, so I was recommended to try and raise an issue here by the node-red forum, since this seems to be a problem with the modbus read node

you can see that the configurations between the modbus read and the modbus-serial read are the same, so it is very weird why the modbus read doesnt work now that i changed to RTU

https://discourse.nodered.org/t/can-connect-to-modbus-rtu-slave-with-modbus-serial-but-not-the-modbus-package/42458

[image]
[image]

**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.com/supporter.html), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (x ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 

   
### Your Environment
**Please tell us about your environment:**

  * (x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :  MAGNA3
     - Version:
     - Vendor : Grundfos 
     - Link   : https://www.grundfos-eica.com/
    
* **node-red-contrib-modbus version:** v5.13.3

* **Node:** (min. 10.x)
   `node --version` = v10.21.0

biancode, 2021-03-29:
Hi @tproulund this is normaly restricted by the device not by Node-RED. The vendor could give you more information about that. If you know the max. to read, then you can read them block by block and add them to the Node-RED context as one block. Later it is possible to get the information from the context as a 60+ Registers-block.

tproulund, 2021-03-29:
Hi @biancode 

Hmm i dont think it is on the device end, as the modbus-serial has no issues reading 62 addresses in one msg from the device. 

I think the device end would also give an error (indicated by a LED) if I was sending a unsupported request.

Could there be some configuration or something causing this issue ?

biancode, 2021-03-29:
Which version do you use to read 62 and are the data of 50+ valid? As I know it was restricted to 50 Registers to read at once.

tproulund, 2021-03-29:
I just tested now to confirm, but yes with modbus-serial and data past the 50 register is still OK and the reading even at 5 sec interval works fine

for the version, here is a snip 

[image]

Here is the specific configuration - I've disabled the modbus-serial configuration, because it somehow blocks your modbus read from connecting to the device, previously i had completely deleted the modbus serial configuration 

`[{"id":"dc2267bc.5ce0b8","type":"debug","z":"dff53b0e.d3f2c8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":1650,"y":780,"wires":[]},{"id":"50fa9411.6970cc","type":"modbus-read","z":"dff53b0e.d3f2c8","name":"","topic":"","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"17","dataType":"HoldingRegister","adr":"300","quantity":"62","rate":"1","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"276147dc.e6e7f8","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"x":1460,"y":780,"wires":[[],["dc2267bc.5ce0b8"]]},{"id":"1c27c187.540dbe","type":"modbusSerial in","z":"dff53b0e.d3f2c8","d":true,"port":"99f00951.c27168","slaves":"17","start":"300","count":"62","dtype":"holding","period":"5","topic":"","name":"","x":1460,"y":600,"wires":[["657ed9c8.b5d698","f67561bc.2d46c"]]},{"id":"657ed9c8.b5d698","type":"debug","z":"dff53b0e.d3f2c8","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":1750,"y":600,"wires":[]},{"id":"f67561bc.2d46c","type":"function","z":"dff53b0e.d3f2c8","name":"Modbus Serial Reader","func":"// The purpose of this block is to make it use to extract information \n// from the modbus read output block which covers everything from\n// Address 301 to 362 with respect to the MODbus functional profile\n\n\n// Firstly, define the MODbus address you are interested in as stated\n// in the functional profile\n// example if you want to read the pump head on address 301\n// then modbus_add = 301 \nvar Modbus_add = 357\n\n// Some addresses are a 32bit, which is indicated by the functional \n// profile, as having a HI and LO part on seperate addresses\n// Example is Opperation time, which is on address 327 and 328\n// if you want to read one of these 32bit addresses, change the \n// Double_add from \"false\" to \"true\". The address should be the first\n// of the double, so from previous example, Modbus_add = 327\n\nvar Double_add = true\n\n// the code below will now extract and format the data point you have \n// selected. you will not need to change anything below\n\nvar index = Modbus_add - 301\nvar msg1 ={'payload' : {'data' : [],'buffer': []}}\n\nif (Double_add === true){\n\n//msg1.payload.data[0] = msg.payload.data[index]\nmsg1.payload.buffer[0] = msg.payload[index*2]//\nmsg1.payload.buffer[1] = msg.payload[index*2+1]//\nindex = index+1\n//msg1.payload.data[1] = msg.payload.data[index]//PowerLO\nmsg1.payload.buffer[2] = msg.payload[index*2]//\nmsg1.payload.buffer[3] = msg.payload[index*2+1]//\n\n// Create new Buffer based on array bytes\nconst buf = Buffer.from(msg1.payload.buffer);\n\n// Represent these bytes as 32-bit unsigned int\nconst value = buf.readUInt32BE();\n\n// save the value\nmsg1.payload.data[0] = value;    \n}\nelse {\n//msg1.payload.data[0] = msg.payload.data[index]\nmsg1.payload.buffer[0] = msg.payload[index*2]//\nmsg1.payload.buffer[1] = msg.payload[index*2+1]//\n\n// Create new Buffer based on array bytes\nconst buf = Buffer.from(msg1.payload.buffer);\n\n// Represent these bytes as 32-bit unsigned int\nconst value = buf.readUInt16BE();\n\n// save the value\nmsg1.payload.data[0] = value;    \n}\nreturn msg1;","outputs":1,"noerr":0,"x":1780,"y":700,"wires":[["b0784453.aa1ef8"]]},{"id":"b0784453.aa1ef8","type":"debug","z":"dff53b0e.d3f2c8","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":2030,"y":700,"wires":[]},{"id":"276147dc.e6e7f8","type":"modbus-client","z":"","name":"RTU MAGNA3","clienttype":"serial","bufferCommands":false,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB0","serialType":"RTU","serialBaudrate":"19200","serialDatabits":"8","serialStopbits":"1","serialParity":"even","serialConnectionDelay":"100","unit_id":17,"commandDelay":1,"clientTimeout":1000,"reconnectOnTimeout":true,"reconnectTimeout":2000,"parallelUnitIdsAllowed":false},{"id":"99f00951.c27168","type":"modbusSerialConfig","z":"","d":true,"port":"/dev/ttyUSB0","baud":"19200","data":"8","parity":"even","stop":"1","name":"Test MAGNA3 RTU"}]`

biancode, 2021-08-12:
RTU-BUFFERED is mostly preferred - RTU is more for RTU Gateways in between, otherwise it needs buffered. On RTU Gateways the buffer is done by the Gateway.

closed 2021-08-12 by biancode