BiancoRoyal/node-red-contrib-modbus #244: Modbus connection of SMA inverter throws error: "Client Not Ready To Read At State init"

robsloetjes, 2021-04-06:
I have a problem with connecting to the modbus of a SMA Sunnyboy 3.0 inverter. 

**Current behaviour:**
When trying to connect to the modbus using a Modbus Read or Modbus Getter nodes I get "Error: Client Not Ready To Read At State init"

 **Expected behavior**
Response with content of the modbus registers.

**Please tell us about your environment:**
I run node red 8.2.1 on a RPI 4 with Home Assistant OS (2021.3.4 / supervisor-2021.03.9). The inverter is a SMA Sunnyboy 3.0 inverter. I can read the modubs with Radzio! Modbus Master Simulator as described on [https://www.sma-sunny.com/en/how-to-test-the-connection-to-your-sma-inverter/](the website of SMA), so it seems that the inverter settings are all right.

This is my config of modbus client
[image]

This is my config of a modbus read node. I tried multiple polling times / timeouts / startup delays
[image]

biancode, 2021-04-08:
hi, please use Nodejs 10 if you have 8 in use and reinstall it. Let me know if that works for you!

robsloetjes, 2021-04-09:
Hi there, my Node Red version (1.2.9, the version 8.2.1 I mentioned earlier turned out to be the add-on version in Home Assistant) uses Node.js version 14.16.0. Could this be the problem?

biancode, 2021-04-10:
see https://github.com/BiancoRoyal/node-red-contrib-modbus  v14 should work - I'll test it asap.

>     Node.JS 14.x is beta supported with v5.x
>     Node.JS 12.x is supported with v5.x
>     Node.JS 10.x is supported with v5.x

biancode, 2021-04-10:
The message "Client Not Ready To Read At State init" means you read maybe to fast after Init? It needs some time in the physical world to connect - some devices need 1 sec. to be ready for the first read. You could use the Status node to handle your read requests. It is just an information to you, that you loose requests in that moment or that your request behaviour is not well designed in the flow.

robsloetjes, 2021-04-10:
Thanks for your reply. 

Still no luck: the status node shows the same error, with no further information. To be clear, I get no output from the modbus read node at all (besides the error). I use a start delay of 10 seconds, that should be enough, isn't it? 

It is strange that on another RPi the node runs succesfully with exact the same settings. The only difference is (as far as I can see) that it runs Node.js v10.24.0

balajitekkiam, 2021-04-18:
> Thanks for your reply.
>
> Still no luck: the status node shows the same error, with no further information. To be clear, I get no output from the modbus read node at all (besides the error). I use a [...]

@robsloetjes @biancode Would you say that downgrading node.js to LTS v10.x will solve the issue?

josepj1986, 2021-04-18:
Hello good morning, 

when you stay in that state it is because normally you are not attacking Modbus addresses well.  SMA has a peculiarity that does not work in the application layer at the communications level, when you access a input address 30513 of the input register, try attack the holding register accessing the area of ​​43513

robsloetjes, 2021-04-20:
> @robsloetjes @biancode Would you say that downgrading node.js to LTS v10.x will solve the issue?

Possibly, but since i run NodeRed in Home Assistant, I can't downgrade tot v10.x.

robsloetjes, 2021-04-20:
> Hello good morning,
>
> when you stay in that state it is because normally you are not attacking Modbus addresses well. SMA has a peculiarity that does not work in the application layer at the [...]

Good evening! My knowledge of modbus is quite limited, but i tried accessing area 43513, but that address is not valid according to a warning. 

And still: on another RPi I can read data from address 30513 with exact same settings. So I think your solution shouldn't need to solve the problem. In the meantime, I push the data from that other RPi with MQTT to Home Assistant. That does the job, but if someone has a solution, let me know ;-)

balajitekkiam, 2021-04-21:
For me, the modbus node functioning properly is critical. So I used the node-red API to reload the flow if the modbus node cannot recover. For me, this solves the issue.

biancode, 2021-08-09:
#211 

The signal run time of SMA has sometimes a maximum of 200 ms.
Maybe you can just request every 250ms on that or better each second. Try to reduce your communication to the Modbus device. Some devices will close stress polling connections on the server side. ;)

The reaction time of the Modbus interface is 5 to 10 seconds. That means you need longer timeouts for that device.
Check the Modbus-Interface description if available or request it from the vendor to get details for a stable communication.
Or switch off the reconnect on timeout option for testing.

en: https://files.sma.de/downloads/FSC11_Modbus-TB-en-10.pdf
de: https://files.sma.de/downloads/FSC11_Modbus-TB-de-10.pdf

rucksman, 2021-08-21:
I did set "Reconnect-Timeout (ms)" to 12000 and "Queue Sendeverzögerung (ms)" to 300 now. The parameters mentioned in the respective document from SMA:

> The signal runtime of the SMA product with Modbus is usually 100 ms.

> For system stability reasons, the time period between data transfers via the Modbus protocol must
be at least ten seconds.

> The reaction time of the Modbus interface is five to ten seconds.

I am not really sure about the difference between reaction time and data transfer time.

closed 2022-08-31 by github-actions[bot]