BiancoRoyal/node-red-contrib-modbus #246: Modebus-Flex-Getter Node stops working every few hours - "initialize" Reopened

rucksman, 2021-04-10:
**I'm submitting a ...**  (check one with "x")

## Simple question to get some hint
- (x ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

 **Current behavior**
As I cannot reopen #245 I create this new issue.

My Flex-Getter node stops working every few hours, sometimes after one or two days. It then only says "initialize". The problem has to be within node-red-contrib-modbus as a php script which logs data into a database from the same modbus device works without problems.

The workaround/solution is:
- log into Node Red
- open the flow
- open the flex-getter node
- open the modbus server
- change some small detail (usually "Queue delay (ms)" one ms up or down)
- update modbus server with this new setting
- close flex-getter node
- deploy
- Flex-Getter node immediately changes it's state to "active" and data is coming in.

So if there is no approach to avoid flex-getter to stop working: Is there a way to automatically "restart" the connection like I always do manually?

 **Expected behavior**
I would expect that the output of data does not stop.
 
 **Minimal reproduction of the problem with instructions**
My flow looks like this:
Flow 1: 
[image]
Read node with one address, interval 10 seconds Screenshot sows the faulty state ("initialized"). Please ignore the blue circles, I just renamed the labels for the screenshot. 

Flow 2;
[image]
It waits until the function node behind the read node from flow 1 has finished, waits another 2 seconds. Then the function node feeds in data to the flex getter node (one address, quantity 96). Complete node and delay node where used to prevent to many or too quick reads. Screenshot also shows the faulty state.
   
### Your Environment
**Please tell us about your environment:**

  * ( x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
I do not have modbus-serial installed, nevertheless it works?
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    - (x) Linux: Kernel 5.10.11
   
   * Description of the connecting Modbus system:
   
     - Name: Solar inverter SMA SB3600SE
     - Vendor: SMA
     - Link: https://files.sma.de/downloads/SMA_Modbus-de-en_V25.zip, https://files.sma.de/downloads/SB5000SE-DEN1618-V11web.pdf
    
* **node-red-contrib-modbus version:** 5.13.3

* **Node:** (min. 10.x)
   `node --version` = 14.15.1

biancode, 2021-11-13:
Maybe someone from the community can test more with SMA here and give more hints and/ a PR.

YoannLe, 2022-01-07:
Hi, same problem with SMA SB5.0, nothing in node-red debug or in docker logs.

Sometimes it takes some days to occur, something some hours...
Just need to restart flows, can't understand what is wrong

YoannLe, 2022-01-12:
With verbose on logs, I get:
```
nodered             | 6 Jan 19:42:59 - [info] Started flows
nodered             | 6 Jan 19:42:59 - [info] [mqtt-broker:mosquittodocker  cerebro] Connected to broker: mqtt://mosquitto:1883
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:30:43 - [error] [modbus-flex-getter:SMA] Timed out
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [warn] [modbus-flex-getter:SMA] Modbus Failure On State sending Get More About It By Logging
nodered             | 10 Jan 00:31:12 - [error] [modbus-flex-getter:SMA] Port Not Open
nodered             | 10 Jan 00:31:42 - [error] [modbus-flex-getter:SMA] Error: Client Not Ready To Read At State init

nodered             | 10 Jan 11:58:00 - [error] [modbus-flex-getter:SMA] Error: Client Not Ready To Read At State init
nodered             | 10 Jan 11:58:14 - [error] [modbus-flex-getter:SMA] Error: Client Not Ready To Read At State init
nodered             | 10 Jan 11:58:44 - [error] [modbus-flex-getter:SMA] Error: Client Not Ready To Read At State init
nodered             | 10 Jan 11:59:11 - [info] Stopping flows
nodered             | 10 Jan 11:59:11 - [info] Stopped flows
nodered             | 10 Jan 11:59:11 - [info] Starting flows
nodered             | 10 Jan 11:59:11 - [info] Started flows

```
SMA Inverters goes in sleep mode the night and shutdown network interface, could it generate this problem?
Why modbus-flex-getter doesn't re connect or reset connection?

Yoann

tug-wv, 2022-02-08:
Have you checked, if the 'Reconnect on timeout' Checkbox is active in the Server Configuration?

YoannLe, 2022-04-18:
My configuration below, sometimes it stop after 2 or 3 days, sometimes one week. 

I may try to make a .pcap and look at the failure in tcp flow. But I think that the root cause is the SMA sleep mode every nights, maybe the reconnect can expire after few tries? But why this is a nonsense when it keep working few days.

closed 2022-08-30 by github-actions[bot]

derjosh76, 2025-11-10:
Maybe useful for others with this issue - my flow was running flawlessly for years, until I updated to v5.45.1. With this update this "initialize" issue occured every few hours.
Now with v5.45.2 it is working for several days now without any occurrences so far.

thx, josh