BiancoRoyal/node-red-contrib-modbus #251: Feature proposal: add before_tx and after_tx events to modbus-client - I will develop and submit PR

Ozzard, 2021-04-28:
**I'm submitting a ...**  (check one with "x")

 - (x) feature request
 
I've also run into the issue that some of the RS485 boards on the RPi need to pull a GPIO pin to switch between TX and RX (see #123 for context).  I'd like to contribute some general code to deal with this, but I want to check three things before I start coding:

* that there isn't an existing way of doing this;
* that my approach is reasonable;
* that my event names are reasonable.

I'd like to add two _synchronous_ events to modbus-client, invoked from all of modbus (flex) read, write, and getters.  By default, both events are blank, but they can take Javascript code in exactly the same way as a Node-RED function node.

The events are:
* before_tx: called immediately before bytes are transmitted. For example, this could be used to assert RTS where required.
* after_tx: called immediately after bytes are received. For example, this could be used to clear RTS where required.

The events should only need access to the modbus-client node, and any return should be ignored,  They should be able to raise errors - otherwise there's no way of knowing that they're not working - but that should be the limit of their interaction with the rest of Node-RED.

The most obvious interface for this in the editor would be to add two tabs after the "Properties" tab of modbus-client, named "Before transmit" and "After transmit", with interfaces very similar to the "Setup" and "Close" tabs of the Node-RED "function" node.

As I say, I'm happy to add this and submit a PR.  Is it needed? Is it reasonable? Are the event names reasonable? What have I missed?

biancode, 2021-04-29:
Hello @Ozzard - I'm sorry, but for me that is not the right way. As you write "in exactly the same way as a Node-RED function node". So why not to set a function node before and after? With the Link Node you will have the before and after once if needed or you could set multiple nodes etc. Why to add this complexity to this node? the before and after is fast enough with the function nodes to react immediately on inputs and outputs to the Modbus nodes. Maybe it makes more sense to contribute with additional nodes like the function node with special features to interact and generalize with the Modbus nodes? Thank you very much for the idea and request!

Ozzard, 2021-04-29:
For the chips that form my use case, RTS needs to be asserted before transmitting, and cleared within a few milliseconds of the transmission finishing.  Crucially, it's impossible to receive bytes over RS485 without clearing RTS, so RTS must be cleared before data can be received via Modbus RTU.

Using a function node before many of the nodes would work to assert RTS.  It wouldn't work for the nodes with built-in polling such as modbus-read, because (as far as I can see) there is no way to trigger a function node before they transmit data on the bus.

Using a function node on node output would _not_ work to clear RTS and allow receiving data.  The timing is wrong; the Node-RED message is sent from the node output after data is received, with a payload of the received data.  But RTS needs to be cleared _before_ data can be received, in order to allow the reception of that data.  So a function node - or anything else using Node-RED's flows - cannot do this.  So this capability has to be added within the node's control flow, once it has finished its transmit, in order to turn the bus round.

Sorry - as a typical developer, my original issue gave a solution without a clear enough explanation of the problem!  Does this now help explain my problem a little more clearly?

biancode, 2021-04-29:
OK understood, then feel free to send your PR your suggestions for the events are fine and let me know if you need some help ;)

biancode, 2021-04-29:
Please, use the develop branch and open a feature-branch from develop if possible!

closed 2022-08-30 by github-actions[bot]