BiancoRoyal/node-red-contrib-modbus #252: Version 6.x with NodeJS LTS 16

biancode, 2021-04-29:
**I'm submitting a ...**  (check one with "x")

## Upgrade
 - (X) upgrade request
 
 **What is the motivation / use case for changing the behavior?**
 
There are more and more issues depending on the new NodeJS versions and the package needs to get an upgrade for the next LTS NodeJS v16 in fall 2021.

* **Node:**
   `node --version` = v16.x

biancode, 2021-04-29:
Dear companies, please read #183

biancode, 2021-11-11:
We will need a bit more time until Jan 2022, because we had a lot issues over the year from Covid.

dankomick, 2022-02-06:
Hi, thanks a lot for your work, will be soon yout next release, ready to work with node.JS 16.x or 17.x?
I was trying very hard to downgrade node.JS version in Termux (android), but I couldn't.

Thanks in advance.
Dankomick.

biancode, 2022-02-06:
@dankomick thank you for the question - we try to support v16.x by default and have v17 in mind, but the actual Covid situation face us with a lot of unexpected issues over the last two month. From that we are really out of range from our release plannings for open source projects at the moment.

dankomick, 2022-02-06:
@biancode So sorry about your unexpected issues.
Waiting for future news, to use your node-red-contrib-modbus again in Termux-android.

Thanks for your reply and specially for your work.

biancode, 2022-02-12:
v5.20+ with node v16.4.0 passes all testings well for now ...

biancode, 2022-02-12:
**I'm submitting a ...**  (check one with "x")

## Upgrade
 - (X) upgrade request
 
 **What is the motivation / use case for changing the behavior?**
 
There are more and more issues depending on the new NodeJS versions and the package needs to get an upgrade for the next LTS NodeJS v18.

* **Node:**
   `node --version` = v17.x

biancode, 2022-02-12:
@dankomick please give v5.20+ a try and share your experience with community here!

dankomick, 2022-02-12:
Hi, sorry, I can’t find any other module version than 5.16.0.

Should I wait for something to download the new version?

 

Best regards.

 

-----------------------------------------------------------

2022-02-12T12:15:49.774Z Install : node-red-contrib-modbus 5.16.0

 

2022-02-12T12:15:48.295Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict ***@***.***

2022-02-12T12:16:17.146Z [err] npm

2022-02-12T12:16:17.147Z [err]  ERR! code EBADENGINE

2022-02-12T12:16:17.165Z [err] npm 

2022-02-12T12:16:17.166Z [err] ERR! engine Unsupported engine

2022-02-12T12:16:17.166Z [err] npm ERR! engine Not compatible with your version of node/npm: ***@***.***

2022-02-12T12:16:17.166Z [err] npm ERR! notsup Not compatible with your version of node/npm: ***@***.***

2022-02-12T12:16:17.167Z [err] npm ERR! notsup Required: {"node":">=10 <15"}

2022-02-12T12:16:17.167Z [err] npm ERR! notsup Actual:   {"npm":"8.1.2","node":"v16.13.2"}

2022-02-12T12:16:17.215Z [err] 

2022-02-12T12:16:17.215Z [err] npm ERR! A complete log of this run can be found in:

2022-02-12T12:16:17.215Z [err] npm ERR!     /data/data/com.termux/files/home/.npm/_logs/2022-02-12T12_16_17_177Z-debug.log

2022-02-12T12:16:17.266Z rc=1

Enviado el: sábado, 12 de febrero de 2022 11:10
Para: BiancoRoyal/node-red-contrib-modbus ***@***.***>
CC: dankomick ***@***.***>; Mention ***@***.***>

biancode, 2022-02-12:
@dankomick it should be available soon - libraries.io show the right version and I made the update request to node-red today
[image]

dankomick, 2022-02-14:
Feedback update: 
- node-red-contrib-modbus installed ok on Latest LTS Version: 16.14.0 (includes npm 8.3.1), windows 8.1 machine
- modbus-read module working perfect for me, reading from USB to RS485 adapter (chip CP2102): 

Thanks for all your efforts @biancode .
Best regards.

closed 2023-03-15 by github-actions[bot]