BiancoRoyal/node-red-contrib-modbus #253: using the high queue reset stops the polling.

ouinouin, 2021-05-06:
## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial]
 **Current behavior**
while resetting the queue (via code provided as in the documentation to do so) , my flex getter stops sending the values, there is no more output from the getter. the getter is still marked as active with a small green square below it , but doesnt go to the other states that you see when its polling.
the issue is with a setup with serial , ut due to the nature of the issue (the queue) i cannot reproduce with modbus contrib serial.
 
 **Expected behavior**
I expect the reset queue not to stop the polling, i suspect either i use incorrctly the queue, or there is somethign to thrigger on the getter to restart normal operations after the queue being full ?  
 **Minimal reproduction of the problem with instructions**
 
   1. run the flow and choose an update rate that the slave receiver cannot follow to ensure the queue will build up 
   2. setup a small queue (10 or 20 messages in the queue) + a  catch node catching messages from the queue :
 ```
  (if("high level reached" === msg.state) {
    msg.resetQueue = true;
    return msg;
}
```
[debug modbus serial queue.zip](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/6434286/debug.modbus.serial.queue.zip)

   3. observe the polling stopped when the msg.reseQueue is sent to the queue node.
   
### Your Environment
**Please tell us about your environment:**

modbus serial with high quality usb to RS485 adapter (prolific adapter sold by radiospare)
  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  
  * Device :
 
    - (x) Linux: v 
   
   * Description of the connecting Modbus system:
   
     - Name   : eastron 
     - Version: SDM120
     - Vendor : eastron 
     - Link   : https://
    
* **node-red-contrib-modbus version:** v 5.14.0
*  nodered latest docker , V1.2.9.

biancode, 2021-08-08:
Hi @ouinouin, did you fill in the msg.unitId on resetQueue true? It could be a problem on parseInt of the unitid in that case.

biancode, 2021-08-09:
new Test on that from next version 5.15+

commit 9ad45314 (2021-11-13): fix: #253 queue info reset
closed 2021-11-13 by biancode