BiancoRoyal/node-red-contrib-modbus #255: Server node handles reconnect flex-server doesn't

code4me-xx, 2021-05-07:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 

Using an external Modbus sleve tool to connect to the server examples. At disconnect/reconnect the flex-server nodeds won't respond but the server nodes will..
 
 **Expected behavior**
 
 
The flex-server should accept disconnect/reconnect

 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (X) Linux: Mint 19 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : modpoll and others as well
     - Version:
     - Vendor :
     - Link   : https://www.modbusdriver.com/modpoll.html
    
* **node-red-contrib-modbus version:** v5.1.4.0

* **Node:** (v14.x)
   `node --version` = v12.20.2

GlassBang, 2021-05-27:
Hello,
I'm encountering the same problem as you, trying to use the flex server node. Did you find any way to make it works ? I was supposed to automatize a communication between sensors from sigfox and the send the datas through a modbus server to a wit redy automaton. Each time I deploy my flows, the automaton try to reconnect but can't.
If there is no way, would you have an idea on how to use modbus server node ? As there is no unit-Id, I'm not sure if it can be use in the same way than modbus flex server.
Sorry for my lame english, I'm not really familiar with it. I hope that it was still understandable.

code4me-xx, 2021-05-27:
Hi,

I ended up using a function node to include the modbus-serial NPM package directly. Then instantiate variables from the examples in the 'on start' as context, delete them (calling  close() on the server) in 'On Stop'.

It works ok, for now.

Regards Peter

biancode, 2021-08-09:
should be done from PR in 5.14.1

biancode, 2021-11-13:
reopen if there are more issues

closed 2021-11-13 by biancode