BiancoRoyal/node-red-contrib-modbus #259: Reading an address most often returns wrong value (0x00)

J4nsen, 2021-05-28:
## Problem

 - (X) unexpected behavior
 
 **Current behavior**
I want to read some values of my solar inverter (SMA STP 20000TL-30), e.g., the produced kWh so far, but I'm having problems getting the correct value for the LSBits (buffer [7] in this case). In most cases it is zero, but sometimes it contains the correct value. 

I have a working setup with modbustcp, but it has other problems, which I hope to circumvent with this node. 
 
 **Expected behavior**
 
 The buffer should contain the correct value.

 **Minimal reproduction of the problem with instructions**
 
   1. Create a Modbus Flex Getter
   2. Inject `{"value":"Gesamtertrag","fc":3,"unitid":3,"address":30513,"quantity":4}` as msg.payload
   3. Get wrong values (In most cases `0x0f00` instead of `0x0fba`. It rarely works and I'm testing this at night, hence the value should not change)
  
I've attached a wireshark pcap with 3 reads of the address ([modbus.zip](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/6563707/modbus.zip)). The first one is with modbustcp, the other ones with modbus-flex-getter. As far as I can tell the inverter always sends the correct value: `0x0fba`. This is also visible in this screenshot:

[image]

Any ideas and help would be very much appreciated. There is also the possibility that I'm doing something completely wrong. :D

### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it

  
  * Device :
    - (X) Linux: Ubuntu 18.04
   
   * Description of the connecting Modbus system:
   
     - Name   :  Tripower STP-20000TL-30
     - Version: 2.83
     - Vendor : SMA
     - Link   : https://files.sma.de/downloads/MODBUS-HTML_STPTL-30_028303R_V10.zip
    
* **node-red-contrib-modbus version:** v5.14.0

* **Node:** (v14.x)
   `node --version` = v14.17.0

biancode, 2021-07-11:
Hi @J4nsen, the problem should be the same in the modbus-serial lib, too. Please, could you test this with a tiny script and modbus-serial in plain JS or TypeScript? If it is the same problem, then open an issue to the library or test with the newest version if it helps to update to a newer version of modbus-serial for this issue.

biancode, 2021-08-09:
also SMA #244

biancode, 2021-08-09:
The signal run time of SMA has sometimes a maximum of 200 ms.
Maybe you can just request every 250ms on that or better each second. Try to reduce your communication to the Modbus device. Some devices will close stress polling connections on the server side. ;)

The reaction time of the Modbus interface is 5 to 10 seconds. That means you need longer timeouts for that device.
Check the Modbus-Interface description if available or request it from the vendor to get details for a stable communication.
Or switch off the reconnect on timeout option for testing.

en: https://files.sma.de/downloads/FSC11_Modbus-TB-en-10.pdf
de: https://files.sma.de/downloads/FSC11_Modbus-TB-de-10.pdf

biancode, 2021-08-09:
also #211

kristian, 2025-01-08:
I can still reproduce this issue on my SMA inverter. Took me quite some time to corrleate this, however the behavior that I am observing exaxtly matches what was reported here:

- When reading the register with any tool (I used QModMaster) the value is read correctly e.g. a single register `[ab cd]`
- When reading it via Node-RED the least significant byte is (mostly) missing / 00, so `[ab 00]` leading to wrong values reported

I was once able to get the right value right after a restart of Node-RED, but in 99% of cases I get the wrong values as described above. As this is quite an old thread, has there been some kind of solution for this, or a workaround / explanation why this could be happening? Thanks!