BiancoRoyal/node-red-contrib-modbus #262: Modbus Flex Getter removes "parts" property of a message

Girgitt, 2021-07-06:
**I'm submitting a ...**  (check one with "x")

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 In version 5.14.0 when Modbus Flex Getter receives msg which was split by a split node it returns message without parts property which makes it impossible to automatically join messages by the join node.

 **Expected behavior**
 
Behavior from version 5.4.0 is expected so the parts property is not removed. 
I suspect the behavior was changed by this commit: https://github.com/BiancoRoyal/node-red-contrib-modbus/commit/74271d3be55f4719e83e8a044b956be64c20ca57 where function creating a response message was changed
 
 **Minimal reproduction of the problem with instructions**
 
   1. create msg with payload 
        "1":"foo"
         "2": "bar" 
   2. inject the msg to a split node and split by \n for fixed length 1
   3. connect split node to change node and set payload to a valid modbus request
       msg.payload = { 
        'value': 0, 
        'fc': <fc>, 
        'unitid': <unit_id>, 
        'address': <address> , 
        'quantity': <quantity> };   
  4. connect split node to Modbus Flex Getter
  5. connect flex getter to join node with automatic join action and output to debug node
  
   
### Your Environment
**Please tell us about your environment:**

  I have installed the package using node-red built-in nodes palette manager (which installed version 5.14.0 - confirmed using npm list). Then I checked other installation not having described issues and downgraded the package by going to a directory where affected node-red instance keeps configuration files and issuing "npm uninstall node-red-contrib-modbus" followed by "npm install node-red-contrib-modbus@5.4.0" which brought back the previous, correct behavior. 
   
  I am not using modbus serial at all, only modusTCP client via Modbus Flex Getter. 
  
  * Device :
    
    - ( ) macOS: v 
    - ( x ) Linux: Centos 7.9 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name : 
     - Version:
     - Vendor :
     - Link   : https://
   
   ISSUE IS NOT RELATED TO MODBUS PART only to integration with Node-Red.
  
* **node-red-contrib-modbus version:** v5.14.0

* **Node:** (v12.16)
   `node --version` = v12.16.1

biancode, 2021-07-11:
Hi @Girgitt, use the 'keep msg properties' option, please!

[image]

[image]

[image]

Girgitt, 2021-07-20:
Thank you! You have put so much functionality into this node that I missed this change. 

Also your client is so stable sitting quietly and doing its job for months without a glitch I had no reason to closely follow the changelog. It's simply a piece of great work!

closed 2021-07-20 by Girgitt