BiancoRoyal/node-red-contrib-modbus #267: Buffer address offset hard-coded to 8

electromaticboy, 2021-08-16:
**I'm submitting a ...**

## Problem
 - (x) unexpected behavior
 
 **Current behavior**
 When polling multiple holding registers, the reply buffer is filled with register 1 (WORD) going to buffer addresses 0 and 1 (BYTES), register 2 however goes to addresses 8&9, register 3 to 16&17 and so on. That's not consistent with itself (why doesn't register 1 have the offset as well?) and doesn't obey to any rules (a holding register is always 16-bit and multiple registers are used to display alternative data types, not offsets).
 
 **Expected behavior**
 Register 1 mapping seems correct, register 2 should go to buffer[2] and buffer[3], register 3 to buffer[4] and buffer[5], etc.
 
 **Minimal reproduction of the problem with instructions**
The line in https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/master/src/core/modbus-server-core.js
de.biancoroyal.modbus.core.server.bufferFactor = 8
doesn't have any comments and if it has any use, it should be a parameter available in the interface of node red and not hard-coded.

biancode, 2021-08-17:
Hi @electromaticboy, thanks for the question - but that is how the buffer works in nodejs (offset of 8). I think it is more flexible if you can use the nodejs buffer functions to work with the buffer. The registers you expect are on output 1, the buffer is on output 2, isn't it?

biancode, 2021-08-17:
The pure buffer is to implement your own registers size of 8 or 32 or what ever is needed. ;)

electromaticboy, 2021-08-18:
Hi @biancode , I've seen the behavior using the Modbus Flex Server example so each of the four output buffers correspond to a type of register (holding, input, coil, digital). It's just the discrepancy between the first holding register in the buffer (at buffer[0] and buffer[1]) and the rest of them having the offset of 8. It would make sense to have the same offset for all and then yes, one could implement a custom register size for all registers. I've switched to individual get-write nodes that are more consistent in that respect.

closed 2021-08-18 by electromaticboy