BiancoRoyal/node-red-contrib-modbus #268: Bit addressing

dpatel2016, 2021-08-17:
Can we have a feature added to use status and control word? The picture shows below I have modbus address of 40001 and it is suppose to be used with the table below. 
[image]

biancode, 2021-08-17:
Hi, some like this https://flows.nodered.org/node/node-red-contrib-bit  or what is the need here? You can read and write the bits inside a word or register.

dpatel2016, 2021-08-19:
Yes, that's it! I thought the current library might have this feature. 
I am communicating to compressors from RPI 4. 

Thank you!

dpatel2016, 2021-08-22:
I was able to read it, but I cannot write to it. Any suggestions?

dpatel2016, 2021-08-22:
[image]
I am trying to write 504, but it says illegal data value

anversoft, 2021-08-23:
hi, try to use FC = 16 Write Multiple Registers to write instead of FC = 6, sometimes the compressor logics (I am a compressor consultant and technician) are quite restrictive regarding the function codes to use.

dpatel2016, 2021-08-23:
I tried that and I get the following error.
Error: Quantity should be less or equal to register payload array length: undefined Addr: 504 Q: 1

anversoft, 2021-08-23:
I have a few questions to ask:

- have you tried to use the address 40504 instead of 504?

- Can you read the word from that address?

- Have you tried to manually change the setting on the compressor and have you verified that the address 504 (or 40504) changes accordingly?

moreover, on those modbus tables there is usually also an explanation of the addresses and how to read or write them, if there is, can you post the screenshot of the page?

anversoft, 2021-08-23:
Or some compressors include a protection logic that blocks the writing of data when (for example) the motor is running or there is a certain state of the compressor that blocks the writing of the word.
Currently the compressor (I assume) is off, right?

dpatel2016, 2021-08-23:
I tried everything you suggested and it worked. 
I found out that to write the address, you need to offset by 1. I had to write the value to 503 instead of 504. I figured this out from reading different addresses. I wrote it using FC = 6. 

Yes to change this setting, the compressors have to be stopped, which they were. 

I was trying to write this to understand the compressor settings, but what I truly want to write is the following first 2 bits. I am still unable to do this. There are no other explanations provided in the manual. I tried sending an array with 16 bits and individual number such as 3 to represent true for bits 0 nd 1, but I do not see anything happen to the compressor. 

[image]

anversoft, 2021-08-23:
what you are trying to modify is a complete word, therefore you have to send the whole word to the compressor with the bits you are interested in set correctly.

- First you have to read the word containing all the control states.
(For Example: you receive the number 229 so:

229 in binary is 1110 0101 ( Last bit n.7 -> 1110 0101 <- First bit n.0)

Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host
Bit 0 - Value 1 : Compressor is Host

anversoft, 2021-08-23:
What address is this Status and Control associated with? If it is possible you can link me the manual so I try to take a look?

dpatel2016, 2021-08-23:
Address is 40001 for writing and 40002 for reading.
[RS220IE Product Manual.pdf](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/7033107/RS220IE.Product.Manual.pdf)

anversoft, 2021-08-23:
I just read that bit 0 must be set to 1 in order to modify the others, try to send 1 as the first number and then modify the others accordingly as it is written in the notes.

dpatel2016, 2021-08-23:
Just to confirm, I should send value 1 to address 40001 and then send 7 to the same address. Correct?

anversoft, 2021-08-23:
no, first you have to send 1 to 40001, then send 3 to 40001 and then 7 to the same address.

anversoft, 2021-08-23:
If you read 40001 what does it give you?

Have you tried sending the numbers with the offset? so instead of address 40001 use 40000.

dpatel2016, 2021-08-23:
When I read 40001, it gives me a value of 6 for the compressors that are running and a value of 0 for the ones that are not running.
I tried writing to both 40000 and 40001.

anversoft, 2021-08-23:
Try setting Host on the compressor panel and then changing the other two bits.

dpatel2016, 2021-08-23:
I finally got it, thank you.
In order to change the Host or start the compressor using Modus, I needed to turn on Communications Control, which can only be changed at the compressor. After that, I am able to send 3 and it starts the compressor. I did not find any of these details in the manual and the company does not want to help since they want us to buy their central system when I can tie everything in my central dashboard. 

Again, thank you!

closed 2021-08-24 by dpatel2016