BiancoRoyal/node-red-contrib-modbus #274: Problem trying to get modbus tcp with UnitID=0

luisgonzalezgonzalez, 2021-09-02:
## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - (x ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

I'm trying to read by modbus a huawei SUN2000 solar inverter. The problem is that this inverter uses UnitId=0.
I tried to read it with QmodMaster from my pc and no problem. But when i try with modbus for nodered it gives "Error: UnitId is valid from msg or node" and i'm not able to read it.

 
   
### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Huawei SUN 2000 2Ktl
     - Version:  n/a
     - Vendor : HUAWEI
     - Link   : https://solar.huawei.com/es-ES/download?p=%2F-%2Fmedia%2FSolar%2Fattachment%2Fpdf%2Fes%2Fdatasheet%2FSUN2000-100KTL-M1.pdf
     - 
* **node-red-contrib-modbus version:** v5.14.1
* 

* **Node:** (v14.x)
   `node --version` = v16.5.0

anversoft, 2021-09-02:
Have you tried to send the request using the Modbus-Flex-Getter node?

https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/577ec7223d1ebf2d0f22338870b2595ce5a1e9d6/src/core/modbus-queue-core.js#L144-L191

Send this to the node:

```
msg.payload = 
  {
      fc : 1, //<- The function
      address : 32260, //<- The start address
      unitid: 0, //<- unitId
      quantity : 2, //<- Quantity
  }
 return msg;
```

Or try to use the modbus-flex-sequencer node, the doc are [here](https://github.com/BiancoRoyal/node-red-contrib-modbus/pull/269/files)

luisgonzalezgonzalez, 2021-09-02:
first of all thanks for helping me.

Yes, i have tried with moddus-flex-getter and i get exactly the same error
in node-red "Error: UnitId is valid from msg or node"
i have tried what you told me "modbus-flex-sequencer and the error is
different "Unit ID Not Valid"
i don't know why the system doesn't like unitid=0 and the unitid for huawei
is 0.
best regards

anversoft, 2021-09-02:
Well.... I found the problem.

https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/577ec7223d1ebf2d0f22338870b2595ce5a1e9d6/src/modbus-flex-sequencer.js#L105-L110

luisgonzalezgonzalez, 2021-09-03:
no, this is not the problem....
if you change this line to allow unitid=0 returns the same error that the
others nodes "Error: UnitId is valid from msg or node"
thanks

biancode, 2021-09-21:
> first of all thanks for helping me. Yes, i have tried with moddus-flex-getter and i get exactly the same error in node-red "Error: UnitId is valid from msg or node" i have tried what you told me [...]

Do you read serial or via TCP? On TCP it makes no difference in the most cases. Via TCP it could also work with another UnitId.

biancode, 2021-11-13:
> Well.... I found the problem @biancode.
> 
> https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/577ec7223d1ebf2d0f22338870b2595ce5a1e9d6/src/modbus-flex-sequencer.js#L105-L110

fixed from v5.15.+

closed 2021-11-13 by biancode

MateRadz, 2022-01-09:
Unitid=0 is not fixed for serial.... getting the same errors "Error: UnitId is valid from msg or node"

commit fccd59be (2021-11-13): fix: #274