BiancoRoyal/node-red-contrib-modbus #276: Cannot Install Modbus module and error log provided

ksks1210, 2021-11-02:
[2021-11-02T01_11_04_824Z-debug.log](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/7457370/2021-11-02T01_11_04_824Z-debug.log)
[2021-11-02T01_53_11_960Z-debug.log](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/7457371/2021-11-02T01_53_11_960Z-debug.log)

**I'm submitting a modules installation issue found at node-red or using npm install. same error log i was received

## Problem
 - ( ) unexpected behavior => cannot download modbus module from npm or node-red module
 
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    - ( ) Windows: Windows_NT 10.0.17763
     
* **node-red-contrib-modbus version:** v5.14.1

* **Node:** (v14.x)
   `node --version` = v16.13.0

gonxal0, 2021-11-04:
Same problem on debian

biancode, 2021-11-13:
It likes you need the C++ runtime to install on your Windows - help is more node-gyp - please ask the Node-RED community for more help!

biancode, 2021-11-13:
We have no Windows - we can just help on Linux and macOS

closed 2021-11-13 by biancode