BiancoRoyal/node-red-contrib-modbus #280: Status change not recognized by status node

awoSYS, 2021-11-25:
**I'm submitting a ...**  (check one with "x")

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
I've got an application with a Modbus Flex Gettern node. From time to time reading fails with an error like `Client Not Ready To Read At State init`. My problem is that the node status seems to stay "active" all the time, even while I get such error messages. I.e. the small square tile below the node stays filled green, text says "active" and a status node observing this Modbus node does not recognize any status change.
 
 **Expected behavior**
 
During read error like `Client Not Ready To Read At State init` the modbus node status should be shown as "init" and a status node should throw a status change event with status "init".
 
 **Minimal reproduction of the problem with instructions**
 
   1. Create flow with a Modbus node and an inject node attached to its input.
   2. Add a status node observing the Modbus node's status and a debug node attached to the status node's output.
   3. Create new Modbus connection (MODBUS_RTU) and link it to a non-existing device, like `/dev/nonexistingtty`.
   4. Deploy.
   5. Modbus node shows "active", even though it cannot be connected, since the device doesn't even exist.
   6. The status doesn't output anything, even though I'd expect something like "init", "connecting", "reconnecting", "broken" etc.
   
### Your Environment
**Please tell us about your environment:**

- Node-RED v2.1.3 (Docker image)

  * (x) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  * Device :
    - Raspbian: 9 (stretch)
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X  (don't know how to determine inside a Docker container, package.json says version latest)

* **Node:** (v14.x)
   `node --version` = v14.18.1

biancode, 2022-02-12:
@awoSYS In your reproducible description it is the serial interface with the OS and the way of "all is a file" - if the operating system opens up the dev with no error, then it is "active" and this is more in the serial lib as in the Node-RED contrib-package. It could be possible to open a none existing dev in the OS, but if you try to do some operations, then it will raise errors. I'll have a look for that in the next version if we can find a better solution for that case.

awoSYS, 2022-08-01:
@biancode Thanks for checking and explaining. Any update so far?

biancode, 2022-08-01:
Hi, did you switch on this option?

[image]

closed 2022-11-01 by github-actions[bot]