BiancoRoyal/node-red-contrib-modbus #283: Reading 5 different devices

dpatel2016, 2021-12-10:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - (x ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
I have 5 compressors connected over ethernet and I am reading 4 different Modbus addresses from them using Modbus flex getter. They work sometimes and sometimes they do not. I mean that they time out and keep reconnecting before they output what they read. These outputs go to a function, but its taking too long for it to read. Is there an easy or efficient way to do this. I am using 5 flex getter for 5 different IPs. 
 
 **Expected behavior**
 
I was hoping the connection would stay alive and the information would be read when the node is injected. 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * (X ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Compresor
     - Version:
     - Vendor : IR
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X

* **Node:** (v14.x)
   `node --version` = vX.X.X

biancode, 2021-12-11:
@dpatel2016 you could share the flow via e-mail with me to check it and why the flex-getter and not just readers for that case? I'd use 5 reader with a client on each. First ry this: check the polling rate and specifications of the Modbus slaves (servers/compressors) if they have limitations for Modbus reads! In the most cases the users try to read to fast or often. Modbus polling each >1sec works mostly fine. If the polling is to fast or often in time, then devices can or will close the connection. Or your chosen timeout is to short. Some devices will answer very slow >2sec - try to give more timeout and poll more slowly. Give the device time >=5sec before you read the first time after connection is done. A delay before start were contributed from us in the inject node for that.

dpatel2016, 2021-12-13:
@biancode I emailed you the flow as an attachment. The compressor manual does not say anything regarding the polling rate.