BiancoRoyal/node-red-contrib-modbus #285: I have node-red v17.3.0 not compatible with the latest node-modbus version

uriva9w, 2022-01-17:
**I'm submitting a unexpected behaviour**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. Try to install node Modbus
   2. Failed becuase it is not compatible with my node-red version
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( x) Other:
         - Name: Termux
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Modbus simulator
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** vX.X.X

* **Node:** (v14.x)
   `node --version` = v17,3,0 npm 8.3.0

biancode, 2022-01-17:
@uriva9w the Readme.md gives an overview, which versions are compatible.

bernardstef, 2022-01-27:
Hi !
Same issue here :

2022-01-27T07:56:57.460Z [err]  Not compatible with your version of node/npm: node-red-contrib-modbus@5.16.0
2022-01-27T07:56:57.461Z [err]  Required: {"node":">=10 <15"}
2022-01-27T07:56:57.461Z [err]  Actual:   {"npm":"8.3.1","node":"v17.4.0"}

Thanks for your hard work ! 👍

biancode, 2022-02-12:
Please, let the community know how it works for you!

uriva9w, 2022-02-12:
Fear Blanco
The very much for Share your work
I tried to install your v5.20 but failed in my Termux environment

uriva9w, 2022-02-12:
Sorry some typos in my latest post
Dear Blanco
Thx very much

closed 2022-02-12 by uriva9w

biancode, 2022-02-12:
@uriva9w you could try to work with npx in Termux - did you test it this way?