BiancoRoyal/node-red-contrib-modbus #287: Problems with Modbus reconnection.

carflomi, 2022-01-29:
## Problem
 - Hello, I know that the node reconnection problem (already known) is solved, is to downgrade from Node.js to V12, but I would like to ask if this way of solving is still the only one, especially thinking about possible incompatibilities of new versions of node-network with Node.js V12.
 
   
### Your Environment
**Please tell us about your environment:**

  * (X ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.14.1

* **Node:** 
   `node --version` = v16.13.1

nygma2004, 2022-03-29:
I would also like to add some observation. I am not sure if this is related to this one or if it should be a separate issue.
I am only using Modbus TCP (x86 based computer, Debian 11, Node-Red 2.1.3, Node.js v14.18.2 - but I noticed the same issue in my RaspberryPI as well with raspbian and older version of Node-Red and Node.js).

If I have some internet outage (internet connection goes down, but my router keeps running, or I need to restart the router) modbus connection is lost. The messages keep piling up in the queue, and even if I send a reset queue message nothing happens. The only solution is to re-deploy the flow. 

I would be happy if you can offer some workaround for this problem that I can incorporate in the Node-Red flow.

nygma2004, 2022-07-31:
Adding a comment to keep this alive. This is still an issue for me.

closed 2022-10-31 by github-actions[bot]