BiancoRoyal/node-red-contrib-modbus #288: Modbus-Flex-Getter skips reading remaining devices on the bus after timeout.

anykeymaster, 2022-01-29:
## Problem
 - ( ) bug report
 - (x) unexpected behavior
 
 **Current behavior**
 
 
 I have few slaves on the same bus with different ID's (i.e. 2, 3, 4, 5, 6, 7). And I poll it using Modbus-Flex-Getter. Flow generates separate requests and sends it to Modbus-Flex-Getter. If all devices are "online" (powered on and able to respond on master request), Modbus-Flex-Getter poll it one-by-one in order requests comes to it and works good. Requests usually comes ascending unit ID. First request for ID2, second for ID3 etc. till the last one. Every request processing right way and on Modbus-Flex-Getter output I get response from each slave.
 
But if one of slaves is "offline" (i.e. powered off), Modbus-Flex-Getter gets timeout error and skips polling remaining slaves.
Let we have slaves with ID's 2, 3, 4, 5, 6, 7, and ID5 is powered down. Than Modbus-Flex-Getter poll in right way ID2, ID3 and ID4, gets timeout on ID5 and after reconnect not poll remaining ID6 and ID7. With next attempt everything will be the same way. So on timeout error on any slave, remaining slaves on the bus with greater ID's left unreadable.
 
 **Expected behavior**
 
 
 It will be great if Modbus-Flex-Getter after gettin timeout error on any request in queue ater reconnection continue poll remaining hosts in request queue. In listed example, after getting timeout error on slave ID5 it should reconnect and try to poll remaining slaves with ID6 and ID7
 
 **Minimal reproduction of the problem with instructions**
 

 [{"id":"1d8a2b68e61b54b1","type":"tab","label":"ModbusTCP Test","disabled":false,"info":""},{"id":"836386c17c131621","type":"config","z":"1d8a2b68e61b54b1","name":"Devices","properties":[{"p":"devices","pt":"flow","to":"{\"Gateways\":{\"First-GW\":{\"Devices\":{\"2\":{\"Name\":\"Host2\"},\"3\":{\"Name\":\"Host3\"},\"4\":{\"Name\":\"Host4\"},\"5\":{\"Name\":\"Host5\"},\"6\":{\"Name\":\"Host6\"},\"7\":{\"Name\":\"Host7\"}}}}}","tot":"json"}],"active":true,"x":140,"y":80,"wires":[]},{"id":"c73ab6dc0ead9fe3","type":"inject","z":"1d8a2b68e61b54b1","name":"Timer","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"30","crontab":"","once":true,"onceDelay":"5","topic":"","payload":"","payloadType":"date","x":130,"y":240,"wires":[["74db661a7620769e"]]},{"id":"74db661a7620769e","type":"function","z":"1d8a2b68e61b54b1","name":"Query fabric","func":"var output = [];\nvar devices = flow.get('devices')\n\nfor (var gateway in devices.Gateways){\n    for (var device in devices.Gateways[gateway].Devices){\n        query = {\n                payload: {\n                    fc: 3,\n                    unitid: parseInt(device),\n                    address: 35,\n                    quantity: 44\n                },\n                \n                gateway: gateway\n                \n        }\n        \n        if(query){\n           output.push(query)\n        }\n    }\n}\nreturn [output]","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":360,"y":240,"wires":[["c22dfc974f5c617c","b1c5b59a6adbb703"]]},{"id":"b1c5b59a6adbb703","type":"modbus-flex-getter","z":"1d8a2b68e61b54b1","name":"Modbus-Flex-Getter","showStatusActivities":true,"showErrors":true,"logIOActivities":false,"server":"9a511f0dcbbc09ab","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":true,"x":650,"y":240,"wires":[["5b40ac6a0342fc17"],[]],"inputLabels":["ReadRequest"]},{"id":"c22dfc974f5c617c","type":"debug","z":"1d8a2b68e61b54b1","name":"Requests","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":570,"y":470,"wires":[]},{"id":"5b40ac6a0342fc17","type":"debug","z":"1d8a2b68e61b54b1","name":"Responses","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":840,"y":470,"wires":[]},{"id":"9a511f0dcbbc09ab","type":"modbus-client","name":"Getter","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","serialAsciiResponseStartDelimiter":"0x3A","unit_id":"1","commandDelay":"1","clientTimeout":"500","reconnectOnTimeout":true,"reconnectTimeout":"1","parallelUnitIdsAllowed":false}]
   
### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: v 9
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
   Various systems, connected via Moxa MGate or via SNR-ERD-3s (so polling makes by Ethernet)
   
     - Name   :  Omix P99-MLA-3-0.5-RS485
     - Version:
     - Vendor : Omix
     - Link   : https://oven.spb.ru/catalog/?sid=7415&id=3256118
    
* **node-red-contrib-modbus version:** v5.16.0

* **Node:** (v14.x)
   `node --version` = v14.18.3

biancode, 2022-02-12:
@anykeymeister try it without the reconnect handling - a Modbus Gateway should be used without auto reconnect and you could try to switch off the queue - some Gateways have there own queue. Let me know how this works for you, please!

biancode, 2022-02-12:
@anykeymeister for TCP Gateways it is possible to try this - with that you will have a queue per UintId and the Gateway with TCP could be able to handle this.

[image]

anykeymeister, 2022-08-09:
Hello!
Sorry for such delayed reply.

I tried all your proposions and get no luck. 

I have few new details:
- first one: flex getter skips one or more slaves after offline one depending connection timeout setting.
- second one:  if before flex getter node installed rate limiter with time more than (slave timeout+reconnection timeout) - it works ok.
Anyway looks like it skips some queue elements..
I tried both with 5.16.3 and 5.23.1 (by Node-Red) versions.

closed 2022-11-09 by github-actions[bot]