BiancoRoyal/node-red-contrib-modbus #291: MODULE_NOT_FOUND on Node-Red 2.2.0 with library version 5.21.0

mainakae, 2022-02-14:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 After updating Node-RED to version 2.2.0 in Home Assistant, the module seems to be broken: all the modules report as "MODULE_NOT_FOUND".
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. Installed latest version of Home Assistant
   2. Installed the Node-Red addon, which installs Node-Red 2.2.0 with the latest nodered-contrib-modbus (5.21.0)
   3. The module won't work straight out of the box
   
### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (X) Linux: v 
    - ( ) Windows: v 
    - (X) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.21.0

* **Node:** (v14.x)
   `node --version` = vX.X.X

otreborsfw, 2022-02-14:
Same error here. This is the error in the log:

14 Feb 20:27:06 - [warn] [node-red-contrib-modbus/Modbus-Write] Error: Cannot find module 'source-map-support'

a-n-lundgren, 2022-02-14:
Same here:

14 Feb 23:42:55 - [warn] [node-red-contrib-modbus/Modbus-Write] Error: Cannot find module 'source-map-support'
Require stack:
- /opt/node_modules/node-red-contrib-modbus/modbus/modbus-write.js
- /opt/node_modules/@node-red/registry/lib/loader.js
- /opt/node_modules/@node-red/registry/lib/index.js
- /opt/node_modules/@node-red/runtime/lib/nodes/index.js
- /opt/node_modules/@node-red/runtime/lib/index.js
- /opt/node_modules/node-red/lib/red.js
- /opt/node_modules/node-red/red.js

a-n-lundgren, 2022-02-14:
Added "source-map-support" to the npm_packages configuration parameter, then got this:

15 Feb 00:01:13 - [warn] [node-red-contrib-modbus/Modbus-Flex-Sequencer] Error: Cannot find module 'line-by-line'
Require stack:
- /opt/node_modules/node-red-contrib-modbus/modbus/core/modbus-io-core.js
- /opt/node_modules/node-red-contrib-modbus/modbus/modbus-flex-sequencer.js
- /opt/node_modules/@node-red/registry/lib/loader.js
- /opt/node_modules/@node-red/registry/lib/index.js
- /opt/node_modules/@node-red/runtime/lib/nodes/index.js
- /opt/node_modules/@node-red/runtime/lib/index.js
- /opt/node_modules/node-red/lib/red.js
- /opt/node_modules/node-red/red.js

Added line-by-line and everything seems to work. Final configuration:

npm_packages:
  - source-map-support
  - line-by-line

biancode, 2022-02-15:
I had this packages global and did not see the problem in the test environment. Fixed with V5.21.2.

closed 2022-04-21 by biancode