BiancoRoyal/node-red-contrib-modbus #292: Modbus RTU connection

AlexSanderSabino, 2022-02-15:
Good afternoon.

I'm carrying out a project where I intend to carry out the control through the Node-Red where there will be a frequency inverter, with this reading the information (voltage, current and frequency of the network), and also performing the writing of values ​​in it, but I come across with this certain problem:
msg : string[60]
"Modbus Failure On State sending Get More About It By Logging"
along with this other problem
msg : error
"Error: Port Not Open" (and in this error the communication port complies with the parameters of the modbus RTU), whoever has any idea or solution to propose to me, I'm grateful.

Note: the inverter is privately manufactured.

Below is the flow in Node-Red:
[{"id":"39e69f4d3f47baa3","type":"tab","label":"Read Modbus","disabled":false,"info":"","env":[]},{ "id":"e0d056fe9bd79ac8","type":"serial in","z":"39e69f4d3f47baa3","name":"","serial":"580af37d2f6afea3","x":70,"y": 20,"wires":[["4b18866f7c3c629d"]]},{"id":"4b18866f7c3c629d","type":"debug","z":"39e69f4d3f47baa3","name":"","active" :true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto ","x":370,"y":20,"wires":[]},{"id":"bbb8365710e7c137","type":"inject","z":"39e69f4d3f47baa3","name" :"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":" ","once":false,"onceDelay":0.1,"topic":"","payload":"900","payloadType":"num","x":650,"y":20," wires":[["ca0b146ad7165da2"]]},{"id":"5a603fa6facfc763","type":"inject","z":"39e69f4d3f47baa3","name":"","props":[{ "p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay ":0.1,"topic":"","payload":"1300","payloadType":"num","x":65 0,"y":100,"wires":[["ca0b146ad7165da2"]]},{"id":"ca0b146ad7165da2","type":"modbus-write","z":"39e69f4d3f47baa3","name ":"Modbus_Address_13","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"HoldingRegister","adr":"13","quantity":"1", "server":"54dde46e.53267c","emptyMsgOnFail":false,"keepMsgProperties":false,"x":840,"y":60,"wires":[[["bd1e9066551706ad"],["50478a9a773af93b"] ]},{"id":"50478a9a773af93b","type":"debug","z":"39e69f4d3f47baa3","name":"","active":true,"tosidebar":true,"console" :false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":1050,"y": 80,"wires":[]},{"id":"bd1e9066551706ad","type":"debug","z":"39e69f4d3f47baa3","name":"","active":true,"tosidebar ":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x" :1050,"y":40,"wires":[]},{"id":"e3ac070794cfd2dd","type":"modbus-read","z":"39e69f4d3f47baa3","name":"Modbus_Read_Volt ","topic":"","showStatusActi vities":true,"logIOActivities":true,"showErrors":true,"unitid":"1","dataType":"HoldingRegister","adr":"0","quantity":"1"," rate":"1","rateUnit":"ms","delayOnStart":true,"startDelayTime":"","server":"54dde46e.53267c","useIOFile":false,"ioFile":"" ,"useIOForPayload":false,"emptyMsgOnFail":false,"x":130,"y":120,"wires":[[["0910ad4e93099f81"],["78074b1a2f211722"]]},{"id":" 0910ad4e93099f81","type":"debug","z":"39e69f4d3f47baa3","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false, "complete":"false","statusVal":"","statusType":"auto","x":350,"y":100,"wires":[]},{"id":"78074b1a2f211722 ","type":"debug","z":"39e69f4d3f47baa3","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false," complete":"false","statusVal":"","statusType":"auto","x":350,"y":140,"wires":[]},{"id":"580af37d2f6afea3" ,"type":"serial-port","serialport":"COM6","serialbaud":"19200","databits":"8","parity":"none","stopbits":"1" ,"waitfor":"","dtr":"none","rts":"none","cts":"none","dsr":"none","ne wline":"\\n","bin":"bin","out":"char","addchar":"","responsetimeout":"1000"},{"id":"54dde46e.53267c ","type":"modbus-client","name":"Modbus","clienttype":"serial","bufferCommands":true,"stateLogEnabled":true,"queueLogEnabled":true,"tcpHost": "192.168.1.31","tcpPort":"502","tcpType":"DEFAULT","serialPort":"COM6","serialType":"RTU","serialBaudrate":"19200","serialDatabits": "8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"1000","serialAsciiResponseStartDelimiter":"","unit_id":"1","commandDelay":"30" ,"clientTimeout":"500","reconnectOnTimeout":false,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]

AlexSanderSabino, 2022-02-15:
[image]

biancode, 2022-02-16:
please try RTU Buffered

AlexSanderSabino, 2022-02-16:
Good Morning.
yes I defined it as RTU-BUFFERD, the errors still persist, at the command prompt, the serial port is in normal operation, displaying the message serial port COM6 opened at 19200 baud 8N1, in this case the serial port of my computer. I would like to know why these bugs occur, since I use the serial in node to know the status of the port where the device is connected for communication.

AlexSanderSabino, 2022-02-16:
I understand, in this case, I reset the values of the serial modbus network and made a simple flow, the bugs disappeared at first, but I still can't read the device variables, the debug node doesn't show me a message. Maybe it could be address error?
For example I want to read a variable with address: 0x00 value in hexadecimal, Id of the equipment is 1 and the quantity is 2, by logic it should display some value. Well I will try again to send the node file again.

AlexSanderSabino, 2022-02-16:
[Flows.zip](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/8079900/Flows.zip)
I am uploading a ZIP file, containing a file in txt format and another in json format.

closed 2022-08-28 by github-actions[bot]