BiancoRoyal/node-red-contrib-modbus #294: MODBUS-Flex-Getter-Node FSM state errors intermittently

AustinEcowerks, 2022-02-21:
**I'm submitting a ...**  (check one with "x")

## Simple question to get some hint
 
 - (X) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 

   
### Your Environment
**Please tell us about your environment:**

  * ( X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    

    - (X ) Other:
         - Name:Opto 22 Groov Epic
         - Version:
   
 **node-red-contrib-modbus version:** v5.16.0

Here is what happens in debug:
`2/8/2022, 4:30:57 PMnode: VFD MODBUS
msg : string[123]
"Client -> fsm failed state after sending Get More About It By Logging Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"

2/8/2022, 4:31:42 PMnode: VFD MODBUS
msg : string[122]
"Client -> fsm broken state after failed Get More About It By Logging Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"

2/8/2022, 4:31:42 PMnode: VFD MODBUS
msg : string[125]
"Client -> fsm reconnect state after broken Get More About It By Logging Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"

2/8/2022, 4:31:44 PMnode: VFD MODBUS
msg : string[97]
"Client -> fsm init state after reconnecting Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"

2/8/2022, 4:24:49 PM node: VFD MODBUS
msg : string[83]
"Client -> fsm init in 2000 ms Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"

2/8/2022, 4:31:46 PMnode: VFD MODBUS
msg : string[125]
"Client -> fsm connected after state opened Get More About It By Logging Serial@/dev/ttySerMod0.0:9600bit/s default Unit-Id: 1"`

AustinEcowerks, 2022-02-21:
it is similar to this thread:
https://github.com/BiancoRoyal/node-red-contrib-modbus/issues/236

AustinEcowerks, 2022-02-28:
Still having this issue, maybe somebody could tell me how to better optimize my flow?
`
https://github.com/AustinEcowerks/Node-Red/blob/main/Flow
`
[image]

[image]

biancode, 2022-04-20:
"FSM failed" is not under normal condition for me - with that message it has a failure

biancode, 2022-04-21:
@AustinEcowerks in the most cases users try to read or write (communicate) directly after the deploy, but in the physical world this doesn't work and produces failures while the flow engine starts. The device has to be connected with state active, what can consume some time depending on the device(s) and the Modbus implementation of the device(s). For that issue we contributed to the Inject node, long time before, to provide some start delay. The best practice here is it to set some second(s) before the first injects will be started.

biancode, 2022-04-21:
see also #244

closed 2022-08-28 by github-actions[bot]