BiancoRoyal/node-red-contrib-modbus #295: "Modbus Failure On State sending" warn due to timeouts

Killerloop2, 2022-03-23:
**I'm submitting a ...**  (check one with "x")

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
When ANY read/write Modbus node timeouts on a request (es. unit-id nonexistent/disconnected/broken) a warn message is written in Node-Red debug log: "Modbus Failure On State sending Get More About It By Logging"
 
 **Expected behavior**
Not getting such a warning that clogs the log and may make the journal of the device grow exponentially.
 
 **Minimal reproduction of the problem with instructions**
   1. Create a Modbus Read (or Flex) node over any serial you have and configure it to query a non-existent device
   2. Enable both "Show Activities" and "Show Errors"
   
### Your Environment
**Please tell us about your environment:**

  * (x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.21.2
* **Node:** (v14.19.0)

biancode, 2022-03-23:
Hi @Killerloop2 it is a log about a failure on sending - means unexpected and warn level for me. It sends out the warning. You could stop sending requests in that case and check the connection state to solve this in your case.

DmitrySidorow, 2022-04-19:
I also don't like constant warnings after each restart of flows. I would like a separate checkbox to disable their output to the console.

biancode, 2022-04-20:
it is open source - feel free to send PRs to improve the logging

biancode, 2022-04-21:
do you have node.showErrors active - this should be set to false to get less logging for that

Killerloop2, 2022-04-21:
the problem is that if you disable showErrors you won't get timeout messages from catch nodes.

closed 2022-08-28 by github-actions[bot]