BiancoRoyal/node-red-contrib-modbus #296: "Error: UnitId is valid from msg or node" using 0 as UnitID with TCP

Kudhoh, 2022-03-26:
- ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - (x) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

Error similar to #274 but the thread says it was fixed. It appears it's not. I'm getting this error while trying to read via TCP with a UnitID = 0. Other UnitIDs give a timeout error. I can confirm using Simple Modbus client that I get responses from the instrument with UnitID = 0 and not with others. I am using the flex getter, but the error is also present with other nodes.
   
### Your Environment
**Please tell us about your environment:**

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (x) Other:
         - Name: Docker Desktop Windows 10
         - Version: Node-RED 2.2.2
   
   * Description of the connecting Modbus system:
   
     - Name : Julabo Magio MS-600F circulator
     - Version:
     - Vendor : Julabo
     - Link   : https://www.julabo.com/fr/produits/cryothermostats/cryostats-circulation/magio-ms-600f
     - 
* **node-red-contrib-modbus version:** v5.21.2

* **Node:** (v14.x)
   `node --version` = v14.18.2

biancode, 2022-04-19:
the default for TCP is UnitId=0 and the check function allows it

[image]

biancode, 2022-04-21:
Did you set unitId = 0 not unitId = "0"?

Kudhoh, 2022-04-21:
Hi,
Thank you for your reply.
I indeed set the unitid as a number, not a string.
`msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 0, 'address': 1 , 'quantity': 1 } 
return msg`

If I use the standard getter or the read node, I get the same error, whether I use 0 as UnitID or leave it blank.

This is the kind of response i'm getting on the modbus output 

`{"_msgid":"af579cf71069b1f5","payload":"","topic":"03ba14958749360f","messageId":"626100631be3e73dc2ab3712","error":"Error: UnitId is valid from msg or node"}`

biancode, 2022-04-21:
found .. next version will get a fix for this

biancode, 2022-04-21:
hm - for serial it has a timeout from the modbus-serial package - tcp works well with zero now (v5.22.+)

```
USB0:115200bit/s default Unit-Id: 0 +0ms
  contribModbus:core:client Client -> {"info":"read msg via Modbus","message":{"unitid":"0","fc":3,"address":"0","quantity":"5","messageId":"626135994062a0db1f2134dd"},"queueUnitId":0,"timeout":3000,"state":"sending"}  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 0 +0ms
  modbus-serial {
  modbus-serial   action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 00 03 00 00 00 05 84 18>,
  modbus-serial   unitid: 0,
  modbus-serial   functionCode: 3,
  modbus-serial   length: 15
  modbus-serial } +52s
  contribModbus:core:client Client -> {"state":"sending","message":"dequeue command done serial","delay":1}  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 0 +1ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 7d7a2c0b954a013c +3s
  contribModbus:core:client Client -> {"info":"queue response activate sending","sendingAllowed":true,"serialSendingAllowed":true,"queueUnitId":0}  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 0 +3s
  contribModbus:queue:core queue serial unlock command node name: undefined id: 7d7a2c0b954a013c +1ms
  modbus-serial modbusErrorHandling:Timed out +48s
21 Apr 12:44:44 - [warn] [modbus-client:7d7a2c0b954a013c] Client -> fsm failed state after sending Get More About It By Logging  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 0
```

Kudhoh, 2022-04-21:
That was fast, thanks. I'll get back to you when I'll update the node.

closed 2022-08-28 by github-actions[bot]