BiancoRoyal/node-red-contrib-modbus #298: Problem with retrieval of Modbus data from serial USB adaptor

FNW2022, 2022-04-21:
**I'm submitting a ...** 

## Simple question to get some hint
 - ( x ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

Retrieval of data from Modbus chip "FT232RL" (0403:6001 - Future Technology Devices International, Ltd FT232 Serial (UART) IC) not possible. It's connected to my system Raspberry Pi 3+ as device "/dev/ttyUSB2".
Each query of data via node type "Modbus-Read" results in timeout error with message "Modbus Failure On State sending Get More About It By Logging".
But when I run this task on a Windows machine using the same USB adaptor, I'm able to receive data as told in specification (except that address offset of 1). See attached screenshot for further details.
[image]
Which steps do I have to perform in Node-RED environment running on Linux system of Raspberry Pi 3+ to access the Modbus data?
 
 **Current behavior**
Use node type "Modbus-Read" in Node-RED environment on Linux system of Raspberry Pi 3+ to query Modbus data from device "/dev/ttyUSB2". It always results in timeout error with message "Modbus Failure On State sending Get More About It By Logging".
 
 **Expected behavior**
I want to be able to receive the Modbus data queried by response being returned.
 
 **Minimal reproduction of the problem with instructions**
Here is the Node-Red flow I actually use to perform the data query:
[{"id": "3bbbf98dd4668f9e","type": "tab","label": "Power meter (MODBUS)","disabled": false,"info": "","env": []},{"id": "b31478d755f84c64","type": "debug","z": "3bbbf98dd4668f9e","name": "","active": true,"tosidebar": true,"console": false,"tostatus": false,"complete": "true","targetType": "full","statusVal": "","statusType": "auto","x": 510,"y": 300,"wires": []},{"id": "003b5f4cf6cfb09b","type": "modbus-read","z": "3bbbf98dd4668f9e","name": "Voltage test","topic": "","showStatusActivities": true,"logIOActivities": false,"showErrors": true,"unitid": "1","dataType": "HoldingRegister","adr": "2","quantity": "1","rate": "10","rateUnit": "s","delayOnStart": false,"startDelayTime": "","server": "de2bb82a043f4623","useIOFile": false,"ioFile": "","useIOForPayload": false,"emptyMsgOnFail": false,"x": 250,"y": 300,"wires": [["b31478d755f84c64"],["b31478d755f84c64"]]},{"id": "de2bb82a043f4623","type": "modbus-client","name": "DC Meter","clienttype": "serial","bufferCommands": true,"stateLogEnabled": false,"queueLogEnabled": false,"tcpHost": "127.0.0.1","tcpPort": "502","tcpType": "DEFAULT","serialPort": "/dev/ttyUSB2","serialType": "RTU-BUFFERD","serialBaudrate": "2400","serialDatabits": "8","serialStopbits": "1","serialParity": "none","serialConnectionDelay": "100","serialAsciiResponseStartDelimiter": "0x3A","unit_id": 1,"commandDelay": 100,"clientTimeout": 1000,"reconnectOnTimeout": true,"reconnectTimeout": 2000,"parallelUnitIdsAllowed": false}]

### Your Environment
  * ( x ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  * Device :
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( x ) Raspbian: Raspbian GNU/Linux 11 (bullseye)
             Kernel version: Linux 5.15.32-v7+ armv7l (on Raspberry Pi 3+)
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
     - Name: FT232RL (USB Full Speed to Serial UART IC)
     - Vendor : Future Technology Devices International
    
* **node-red-contrib-modbus version:** v5.21.2

* **Node:** (v14.x)
   `node --version` = v16.14.2

Thank you very much in advance for your efforts!
Christian Hüller (newbie to usage of Node-RED)

biancode, 2022-04-22:
@FNW2022 please, check the read write access to "/dev/ttyUSB2" if is Node-RED really able to read and write it. Then start Node-RED with:

`DEBUG=contribModbus*,modbus* node-red -v`

and send a zipped log to my e-mail or upload it here if it has no secrets in it

biancode, 2022-04-22:
you set the address to 2 in the test you use 1281 ...

FNW2022, 2022-04-22:
Thank you very much for your fast reply!
You are correct about the different registers: Register 2 contains the voltage applied to the energy meter I am trying to read data from. I tried with this first before reading the date & time registers (1281 to 1286) as they always contain "simple and useful" data, but reading register 2 also works as expected on Windows with _QModMaster_.
Naturally, I also changed the setting of the _Modbus-Read_ node now so it perfectly reflects the settings I used on my Windows machine to get the date & time from the device.

[image]

Unfortunately, using these other (valid) regsiters did not change anything regarding the timeout issue.
I just checked the permissions on the ttyUSB device and did a reboot to be extra sure that nothing is "stuck" (as many people also recommend a reboot in case of permission issues).
After the reboot, my USB to Serial converter "moved" to the first place of the ttyUSB list (so it is now **ttyUSB0**) and it seems to be properly configured in my opinion (see first rows of logs below).

As the console log isn't very large, I decided to attach it here directly and hope you see anything helpful in it:

`pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@raspberrypi:~ $ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Apr 22 11:11 /dev/ttyUSB0

pi@raspberrypi:~ $ node-red-pi --max-old-space-size=256 DEBUG=contribModbus*,modbus* node-red -v
22 Apr 11:15:05 - [info]

Welcome to Node-RED
===================

22 Apr 11:15:05 - [info] Node-RED version: v2.2.2
22 Apr 11:15:05 - [info] Node.js  version: v16.14.2
22 Apr 11:15:05 - [info] Linux 5.15.32-v7+ arm LE
22 Apr 11:15:07 - [info] Loading palette nodes
22 Apr 11:15:11 - [info] node-red-contrib-ccu version: 3.4.2
22 Apr 11:15:12 - [info] Settings file  : /home/pi/.node-red/settings.js
22 Apr 11:15:12 - [info] Context store  : 'default' [module=memory]
22 Apr 11:15:12 - [info] User directory : /home/pi/.node-red
22 Apr 11:15:12 - [warn] Projects disabled : editorTheme.projects.enabled=false
22 Apr 11:15:12 - [info] Flows file     : /home/pi/.node-red/DEBUG=contribModbus*,modbus*
22 Apr 11:15:12 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

22 Apr 11:15:12 - [info] Server now running at http://127.0.0.1:1880/
22 Apr 11:15:12 - [info] Starting flows
22 Apr 11:15:12 - [warn] [modbus-read:Read Time] Client -> open node dec6bc846eb4647aundefined
22 Apr 11:15:12 - [warn] [modbus-client:DC Meter] Client -> fsm init state after new
22 Apr 11:15:12 - [warn] [modbus-client:DC Meter] Client -> first fsm init in 500 ms Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:12 - [info] Started flows
22 Apr 11:15:12 - [warn] [modbus-client:DC Meter] Client -> fsm connected after state opened Get More About It By Logging Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:12 - [warn] [modbus-read:Read Time] Client -> startIntervalReading node dec6bc846eb4647aundefined
22 Apr 11:15:23 - [warn] [modbus-client:DC Meter] Client -> fsm failed state after sending Get More About It By Logging Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:23 - [warn] [modbus-read:Read Time] Client -> resetIntervalToRead node dec6bc846eb4647aundefined
22 Apr 11:15:23 - [warn] [modbus-read:Read Time] Modbus Failure On State sending Get More About It By Logging
22 Apr 11:15:23 - [warn] [modbus-client:DC Meter] Client -> fsm broken state after failed Get More About It By Logging Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:23 - [warn] [modbus-client:DC Meter] Client -> fsm reconnect state after broken Get More About It By Logging Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:23 - [error] [modbus-read:Read Time] Error: Timed out
    at /home/pi/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/core/core/modbus-client-core.js:79:21
22 Apr 11:15:25 - [warn] [modbus-client:DC Meter] Client -> fsm init state after reconnecting Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:25 - [warn] [modbus-client:DC Meter] Client -> fsm init in 2000 ms Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:27 - [warn] [modbus-client:DC Meter] Client -> fsm connected after state opened Get More About It By Logging Serial@/dev/ttyUSB0:2400bit/s default Unit-Id: 1
22 Apr 11:15:27 - [warn] [modbus-read:Read Time] Client -> startIntervalReading node dec6bc846eb4647aundefined
`

FNW2022, 2022-04-25:
Hi again,

I have good news: After I re-installed Node-RED and your Modbus module on a clean current Raspian OS Lite installation, everything works like a charm!
No strange timeouts or other errors, but just a nice set of valid data after each request. :)

Sorry for stealing your time - it seems that I just tripped up my own heels with a messed up Debian installation on my RPi 3+.
I will send you $50 to keep up the good work which helps me a lot in my private smart home project...

Best Regards,

Christian

closed 2022-04-25 by FNW2022