BiancoRoyal/node-red-contrib-modbus #303: crash for non-open port

FStefanni, 2022-06-14:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
Under raspberry, if I set a port that cannot be open, an exception is thrown. Such exception is not catched, and therefore the nodered instance chrases.

 **Expected behavior**
 
 
The exception should be catched, and an error should be reported to the user, by using nodered mechanisms.

 **Minimal reproduction of the problem with instructions**
 
   1. Just configure a not open port.

```
[error] Error: Unhandled error. (SerialPortError {
  name: 'SerialPortError',
  message: 'Port is not open',
  errno: 'ECONNREFUSED',
  stack: 'Error: Port is not open\n' +
    '    at unixWrite (/home/pi/.node-red/node_modules/@serialport/bindings/lib/unix-write.js:25:15)\n' +
    '    at /home/pi/.node-red/node_modules/@serialport/bindings/lib/linux.js:72:7'
})
    at new NodeError (node:internal/errors:371:5)
    at ModbusRTU.emit (node:events:379:17)
    at ModbusRTU._onError (/home/pi/.node-red/node_modules/modbus-serial/index.js:462:10)
    at RTUBufferedPort.emit (node:events:390:28)
    at SerialPort.<anonymous> (/home/pi/.node-red/node_modules/modbus-serial/ports/rtubufferedport.js:77:14)
    at SerialPort.emit (node:events:390:28)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  * (x) I have installed this node locally by running npm install inside .node-red dir.
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (x) Raspbian: v 4
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** v5.21.2

* **Node:** (v14.x)
   `node --version` = v16.13.2

biancode, 2022-07-08:
see #208

biancode, 2022-07-22:
Please, reopen if you see more points in this.

closed 2022-07-22 by biancode

biancode, 2022-07-29:
With v3.0.x of node-red we decided to handle errors in our package from v5.22.x+