BiancoRoyal/node-red-contrib-modbus #304: Codesys-CSV-To-IO example not working

Kortenbach, 2022-06-20:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 I have looked at the CODESYS-CSV-To-IO example. It has some errors in the "filter named register" function (I changed the example documentation here to include a few changes to the code) I'm able now to generate the sdm630.json file but in another folder than in the example (I'm running a docker image). This is where I get stuck. How does Modbus read refer to this JSON file. There is no documentation and the example code doesn't work. Can you give me a few hints on how to proceed?
 
 **Expected behavior**
 I expected a working example on how to map Modbus IO to named IO's
 
 **Minimal reproduction of the problem with instructions**

   1. Download the Codesys-CSV-To-IO example
   2. Apply to code changes I made in the Wiki
   3. The code now works and produces the desired .json file
   4. Cannot proceed from here due to lack of information
   
### Your Environment
Running Node-RED from a docker container under Windows 10.

  * (x) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
    
* **node-red-contrib-modbus version:** latest

* **Node:** (latest)
   `node --version` = vX.X.X

biancode, 2022-08-01:
Some example files are to find in the extras folder.

[image]

biancode, 2022-08-01:
I try to create more docs soon ...

biancode, 2022-08-01:
[Loom IO-File feature overview](https://www.loom.com/share/32a68ae280464000bdc70ade8bb0969f)

closed 2022-11-01 by github-actions[bot]