BiancoRoyal/node-red-contrib-modbus #305: Modbus-Response-Filter doesn't work

Kortenbach, 2022-06-20:
**I'm submitting a ...**  (check one with "x")

## Problem
 - (x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
Node "Modbus-response-filter" issues an error "Error: 0 does not match 8". After "fixing" this error by entering "0" instead of 8 under "Modbus-Response-Filter-->Properties-->Registers" the error disappears but the filter doesn't work. This is because Message.payload always is a zero length array instead of an array with actual values in it.
 
 **Expected behavior**
A working filter based on the names in the JSON file.
 
 **Minimal reproduction of the problem with instructions**
   1. Install Modbus Nodes into Node-red
   2. Create a valid JSON file from your Codesys device CSV (checkout the Wiki for howto and errors in the node example that does this)
   3. Use the Simple-modbus-IO-demo flow to test.
   4. Change the filter to find an existing name 
   5. Check out the errors
   
### Your Environment
Using build-in node-red-contrib-modbus server
Using latest build of node-red-contrib-modbus
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
    
* **node-red-contrib-modbus version:** latest

* **Node:** latest

biancode, 2022-08-01:
Hi, could you send the debug result before the Response-Filter? The error means, that your array should have 8 entries, but it has input via msg with 0 entries.

Kortenbach, 2022-10-01:
Hi biancode,
Sorry for not responding. Your message seems to have slipped by. Meanwhile I stopped using the node and coded my own workaround. I cannot remember all the details of my post, so I can't help you. I'm sorry...

Kortenbach, 2022-12-09:
Hi Biancode,

Closed because it was stale or did you manage to solve the problem?

Best regards,
Bert Kortenbach

biancode, 2022-12-09:
from the stale state and we were waiting for the debug result before the Response-Filter, please ;)

Kortenbach, 2022-12-09:
(From my previous post)"Message.payload always is a zero length array instead of an array with actual values in it"
Maybe you can try it for yourself to see what is going on?

biancode, 2022-12-12:
Modbus IO-File has to be improved and more documented for the next version v6.x - Release is planned to be on the roadmap for Node-RED 4.0 Q2-2023

Kortenbach, 2022-12-12:
Hi Biancode,
That's great! Thanks for picking this up!

closed 2023-03-14 by github-actions[bot]