BiancoRoyal/node-red-contrib-modbus #309: node-red-contrib-modbus on armv6 platform not loading (unsupported?)

belveder79, 2022-07-19:
## Problem
 - I can't install the module on a RPiZeroW which is an ARMv6 architecture
 
 **Current behavior**
- npm install works correctly (also compiles correctly from source), but node-red crashes trying to load the module saying that "armv6" is not supported (not very verbose)
 
 **Expected behavior**
- actually I would expect that the module works fine if compiled from source
 
 **Minimal reproduction of the problem with instructions**
   1. get a dietpi (or whatever) install on a PiZeroW (armv6 platform)
   2. install node-red
   3. run npm install node-red-contrib-modbus (optionally build it from source)
   4. restart node-red
   
### Your Environment
**Please tell us about your environment:**
  * (X ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package
  * ( X) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    - (X) Raspbian: v 
         - Name: Dietpi (a Debian/Raspbian derivate)
         - Version: v8.6.1
  
* **node-red-contrib-modbus version:** v5.21.2

* **node-red:** v2.1.4

* **npm:** 
* `npm -v`= 8.3.0

* **Node:**
   `node --version` = v17.3.0

biancode, 2022-07-22:
Hi, what do you get if you start node-red in the verbose mode?

`DEBUG=* node-red -v`

biancode, 2022-07-22:
Maybe you have to install without serial support - it is optional - and you could check the supported platforms of the library dependencies if there is some, who does not support ARMv6. The serialport lib has normally the most restrictions.

belveder79, 2022-07-25:
Thx for the hints...
I finally figured out that it seems to be an incompatibility with node v16/v17 particularely on armv6. I now start node-red with `--no-expose-wasm` and it works as expected. I found out through [this post](https://github.com/nodejs/node/issues/41402) that it seems to be connected to the dropped WASM support?

biancode, 2022-07-29:
From my perspective node v12 or v14 is maybe good enough for node-red if you have or see no big security issues. see the package.json of node-red for the min. version of node to run node-red. That is mostly also better if you work with serialport. That library is not compatible with all constellations and the lib provides with detailed information about supported node versions and hardware architectures.

belveder79, 2022-07-29:
You are totally right; v12 or v14 certainly suffice; 

the origin of the issue comes from the fact that distros usually move forward with versions for frameworks regardless of incompatibilities of applications like node-red arising from that. 

Still the real problem is that at this point just everything fails for the non-expert in Linux; for the rest of us, it is just tedious to get a particular install of node outside the distro system and fix a few startup scripts, but that's doable and no pain...

dtabone, 2022-09-04:
Nodered v134 running here.. for some reason, everytime i restart nodered, this node disappears and i have to reinstall it. Any idea why this is happening and how i can resolve this?

belveder79, 2022-11-09:
actually after a long story of trial and error, I finally made it work with `--no-expose-wasm`... don't know how long this will work, but thumbs crossed

closed 2022-11-09 by belveder79