BiancoRoyal/node-red-contrib-modbus #316: Wrong reading after a writing on the same branch

wperw, 2022-07-28:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.space/supporter/), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
When I've a Modbus Flex Getter after a Modbus Flex Writer with the Getter and the Writer operating on the same address of the same unitid I'm obtaining a wrong value.
My device report on the upper byte of the register the received writing  on the lower byte of the register, but I get only the lower Byte valoryzed. IfI run 2 separeted branch everything work fine.
 
 **Expected behavior**
 
My device report on the upper byte of the register the received writing on the lower byte of the register.
 
 **Minimal reproduction of the problem with instructions**
 

  ```
[{"id":"084e0351e9d1d38b","type":"tab","label":"Flow 8","disabled":false,"info":"","env":[]},{"id":"cc5ff5269738520b","type":"inject","z":"084e0351e9d1d38b","name":"","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"overrideValue\":[10],\"operation\":\"Resetting\"}","payloadType":"json","x":670,"y":340,"wires":[["e5e99d7cf043e4d1"]]},{"id":"cf4753df3fd2cad6","type":"change","z":"084e0351e9d1d38b","name":"","rules":[{"t":"move","p":"payload.Desc","pt":"msg","to":"writeConf.Desc","tot":"msg"},{"t":"move","p":"payload.Unit","pt":"msg","to":"writeConf.Unit","tot":"msg"},{"t":"move","p":"payload.Range","pt":"msg","to":"writeConf.Range","tot":"msg"},{"t":"move","p":"payload.minDifference","pt":"msg","to":"writeConf.minDifference","tot":"msg"},{"t":"move","p":"payload.enable","pt":"msg","to":"writeConf.enable","tot":"msg"},{"t":"move","p":"payload.overridesByCode","pt":"msg","to":"writeConf.overridesByCode","tot":"msg"},{"t":"move","p":"payload.mustVerify","pt":"msg","to":"writeConf.mustVerify","tot":"msg"},{"t":"move","p":"payload.Type","pt":"msg","to":"writeConf.Type","tot":"msg"},{"t":"move","p":"payload.MaxRetry","pt":"msg","to":"writeConf.MaxRetry","tot":"msg"},{"t":"set","p":"writeConf.fc","pt":"msg","to":"payload.fc","tot":"msg"},{"t":"set","p":"writeConf.unitid","pt":"msg","to":"payload.unitid","tot":"msg"},{"t":"set","p":"writeConf.address","pt":"msg","to":"payload.address","tot":"msg"},{"t":"set","p":"writeConf.quantity","pt":"msg","to":"payload.quantity","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":1210,"y":340,"wires":[["3a63a8c88b850bca","c9372f2d7b0e0438","3c25d964111cd0d7"]]},{"id":"5f066bb106ce5a56","type":"modbus-flex-write","z":"084e0351e9d1d38b","name":"","showStatusActivities":true,"showErrors":false,"server":"c905aad66a977f0a","emptyMsgOnFail":false,"keepMsgProperties":true,"x":1870,"y":340,"wires":[["b21eb001a776f9ef","47cce8885c7bfd81"],[]]},{"id":"4a4b7c1e9d5756c0","type":"debug","z":"084e0351e9d1d38b","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1570,"y":300,"wires":[]},{"id":"e5e99d7cf043e4d1","type":"change","z":"084e0351e9d1d38b","name":"","rules":[{"t":"move","p":"payload.overrideValue","pt":"msg","to":"ovverideValue","tot":"msg"},{"t":"move","p":"payload.operation","pt":"msg","to":"operation","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":840,"y":340,"wires":[["c3c6798ff2f8042d"]]},{"id":"a9b9db1175e2f01e","type":"change","z":"084e0351e9d1d38b","name":"","rules":[{"t":"set","p":"payload.value","pt":"msg","to":"ovverideValue","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":1640,"y":340,"wires":[["026d93f48b393a70","5f066bb106ce5a56"]]},{"id":"b21eb001a776f9ef","type":"debug","z":"084e0351e9d1d38b","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":2050,"y":300,"wires":[]},{"id":"587a245ab1b9dba0","type":"function","z":"084e0351e9d1d38b","name":"Store to context","func":"if (Array.isArray(msg.payload.value))\n{\n    let tmpArr = [];\n    //Non posso sapere se l'ordine è importante, quindi uso un for\n    for(let i = 0; i < msg.payload.value.length; i ++)\n    {\n        tmpArr.push((msg.payload.value[i] & ~msg.verifyFilter) >> msg.rightShiftRequest);\n    }\n    flow.set(msg.payload.unitid +\"_\" + msg.payload.address, tmpArr);\n}\nelse\n{\n    flow.set((msg.payload.unitid +\"_\" + msg.payload.address, ~msg.verifyFilter) >> msg.rightShiftRequest);    \n}\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":2280,"y":340,"wires":[["2e6f05bceda498fa","4e92c17ed09e8b98"]]},{"id":"3e07b658530dde55","type":"modbus-flex-getter","z":"084e0351e9d1d38b","name":"","showStatusActivities":true,"showErrors":false,"logIOActivities":false,"server":"c905aad66a977f0a","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":false,"x":2650,"y":340,"wires":[["cb785d391f960ce5"],[]]},{"id":"cb785d391f960ce5","type":"debug","z":"084e0351e9d1d38b","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":2830,"y":340,"wires":[]},{"id":"47cce8885c7bfd81","type":"change","z":"084e0351e9d1d38b","name":"","rules":[{"t":"set","p":"verifyFilter","pt":"msg","to":"65280","tot":"num"},{"t":"set","p":"rightShiftRes","pt":"msg","to":"8","tot":"num"},{"t":"set","p":"rightShiftRequest","pt":"msg","to":"0","tot":"num"}],"action":"","property":"","from":"","to":"","reg":false,"x":2080,"y":340,"wires":[["587a245ab1b9dba0"]]},{"id":"026d93f48b393a70","type":"debug","z":"084e0351e9d1d38b","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1830,"y":300,"wires":[]},{"id":"3a63a8c88b850bca","type":"delay","z":"084e0351e9d1d38b","name":"","pauseType":"rate","timeout":"1500","timeoutUnits":"milliseconds","rate":"2","nbRateUnits":"3","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"allowrate":false,"outputs":1,"x":1420,"y":340,"wires":[["4a4b7c1e9d5756c0","a9b9db1175e2f01e"]]},{"id":"c9372f2d7b0e0438","type":"function","z":"084e0351e9d1d38b","name":"PrepareContext","func":"flow.set(\"writing\" + \"_\" + msg.payload.unitid + \"_\" + msg.payload.address, {\"operation\" : msg.operation, \"status\" : \"waiting\"});\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1420,"y":300,"wires":[[]]},{"id":"3c25d964111cd0d7","type":"debug","z":"084e0351e9d1d38b","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1390,"y":260,"wires":[]},{"id":"2e6f05bceda498fa","type":"debug","z":"084e0351e9d1d38b","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":2450,"y":300,"wires":[]},{"id":"4e92c17ed09e8b98","type":"function","z":"084e0351e9d1d38b","name":"","func":"let tmpmsg = {};\ntmpmsg[\"writeConf\"] = msg.writeConf;\ntmpmsg[\"ovverideValue\"] = msg.ovverideValue;\ntmpmsg[\"operation\"] = msg.operation;\nmsg = tmpmsg;\nmsg.payload = { value: 0, 'fc': 3, 'unitid': msg.writeConf.unitid, 'address': msg.writeConf.address , 'quantity': msg.writeConf.quantity}\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":2460,"y":340,"wires":[["e3692e45f50fc2bf","3e07b658530dde55"]]},{"id":"08cee1ab2c15bd8a","type":"comment","z":"084e0351e9d1d38b","name":"TODO","info":"Per qualche motivo se leggo in linea i dati sono sbagliati...\nCredo ci sia una bega nell'usere i nodi modbus in cascata.\nSe lo uso staccato, come quello sopra allora funziona tutto...\nCi dev [... cut ...]
  * (x) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
    
* **node-red-contrib-modbus version:** v5.21.5

* **Node:** (v14.x)
   `node --version` = v16.14.2

biancode, 2022-07-29:
Hi, the write node has a variable quantity and for q>1 you have to provide with an array of values [...] and maybe that is the wrong point to send just 0. Otherwise it is too much work for me at the moment to dig deeper, sorry.

see node doc: "For FC 16, msg.payload must be an array[] of comma separated values between 0:65535 each."

biancode, 2022-07-29:
Sorry, it is too much work for me at the moment to dig deeper in your flow. Did you test just write and read in simpler flow? Just write a representing value and read it after write?

closed 2022-10-28 by github-actions[bot]