BiancoRoyal/node-red-contrib-modbus #327: Won't install on node-red docker 3.0.2

pakerfeldt, 2022-09-03:
### Which node-red-contrib-modbus version are you using?

Latest

### What happened?

I'm unable to install node-red-contrib-modbus on the node-red 3.0.2 docker image.
See log:
```sh
$ docker build .
Sending build context to Docker daemon  2.048kB
Step 1/2 : FROM nodered/node-red:3.0.2
 ---> a15fc0f4e930
Step 2/2 : RUN npm install node-red-contrib-modbus
 ---> Running in 573f9e178489
npm notice
npm notice New minor version of npm available! 8.11.0 -> 8.19.1
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v8.19.1>
npm notice Run `npm install -g npm@8.19.1` to update!
npm notice
npm ERR! code 1
npm ERR! path /usr/src/node-red/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! make: Entering directory '/usr/src/node-red/node_modules/@serialport/bindings/build'
npm ERR! make: Leaving directory '/usr/src/node-red/node_modules/@serialport/bindings/build'
npm ERR! prebuild-install warn install No prebuilt binaries found (target=16.16.0 runtime=node arch=x64 libc=musl platform=linux)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.0.0
npm ERR! gyp info using node@16.16.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.10.5 found at "/usr/bin/python3"
npm ERR! gyp http GET https://unofficial-builds.nodejs.org/download/release/v16.16.0/node-v16.16.0-headers.tar.gz
npm ERR! gyp http 200 https://unofficial-builds.nodejs.org/download/release/v16.16.0/node-v16.16.0-headers.tar.gz
npm ERR! gyp http GET https://unofficial-builds.nodejs.org/download/release/v16.16.0/SHASUMS256.txt
npm ERR! gyp http 200 https://unofficial-builds.nodejs.org/download/release/v16.16.0/SHASUMS256.txt
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/src/node-red/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/src/node-red/.cache/node-gyp/16.16.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/usr/src/node-red/.cache/node-gyp/16.16.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/usr/src/node-red/.cache/node-gyp/16.16.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/usr/src/node-red/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! make: printf: Operation not permitted
npm ERR! make: *** [bindings.target.mk:116: Release/obj.target/bindings/src/serialport.o] Error 127
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:527:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 4.15.0-169-generic
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /usr/src/node-red/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v16.16.0
npm ERR! gyp ERR! node-gyp -v v9.0.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /data/.npm/_logs/2022-09-03T13_11_09_142Z-debug-0.log
The command '/bin/sh -c npm install node-red-contrib-modbus' returned a non-zero code: 1
```

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

Create the Dockerfile:
```
FROM nodered/node-red:3.0.2
RUN npm install node-red-contrib-modbus
```
Run:
docker build .

### What did you expect to happen?

Expect node-red-contrib-modbus to be installed on the docker image.

lu4t, 2022-09-04:
Same effect happens if you use node-red-minimal 3.0.2 image.
I am not pasting the error, cause is the same as the one used to open the issue.

biancode, 2022-09-25:
I'd say there something not available in the image to compile serialport.

closed 2022-12-09 by biancode