BiancoRoyal/node-red-contrib-modbus #333: Everything is working but I am logging an error every once in awhile

CTGControls, 2022-09-16:
### Which node-red-contrib-modbus version are you using?

node-red-contrib-modbus = 5.21.2

### What happened?

```json
{
   "topic":"",
   "_msgid":"06c8d264b32ffe05",
   "error":{
      "message":"TypeError: Cannot read property 'disableMsgOutput' of undefined",
      "source":{
         "id":"79f47b97b1b4ffd4",
         "type":"modbus-flex-server",
         "name":"",
         "count":1
      },
      "stack":"TypeError: Cannot read property 'disableMsgOutput' of undefined\n    at Object._inputCallback (/home/pi/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-flex-server.js:140:25)\n    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:210:26\n    at Object.trigger (/usr/lib/node_modules/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)\n    at Object.Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:202:11)\n    at Object.Node.emit (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:186:25)\n    at Object.Node.receive (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:485:10)\n    at Immediate._onImmediate (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:831:52)\n    at processImmediate (internal/timers.js:464:21)"
   }
}
```

### Server

Modbus-Flex-Server Node (Please attach Script to the next section)

### How can this be reproduced?

```json
[
    {
        "id": "79f47b97b1b4ffd4",
        "type": "modbus-flex-server",
        "z": "87e693e9a43e62a9",
        "name": "",
        "logEnabled": false,
        "serverAddress": "0.0.0.0",
        "serverPort": "502",
        "responseDelay": "100",
        "unitId": 1,
        "delayUnit": "ms",
        "coilsBufferSize": 20000,
        "registersBufferSize": 20000,
        "minAddress": 0,
        "splitAddress": 10000,
        "funcGetCoil": "function getFlexCoil(addr, unitID) {\n\tif (unitID === node.unitId && \n\t\taddr >= node.minAddress && \n\t\taddr <= node.splitAddress) { \n\n\t\treturn node.coils.readUInt8(addr * node.bufferFactor) \n\t}  \n}",
        "funcGetDiscreteInput": "function getFlexDiscreteInput(addr, unitID) {\n\taddr += node.splitAddress\n\tif (unitID === node.unitId && \n\t\taddr >= node.splitAddress && \n\t\taddr <= node.splitAddress * 2) { \n\n\t\treturn node.coils.readUInt8(addr * node.bufferFactor) \n\t}  \n}",
        "funcGetInputRegister": "function getFlexInputRegister(addr, unitID) { \n\tif (unitID === node.unitId && \n\t\taddr >= node.minAddress && \n\t\taddr <= node.splitAddress) { \n\n\t\treturn node.registers.readUInt16BE(addr * node.bufferFactor)  \n\t} \n}",
        "funcGetHoldingRegister": "function getFlexHoldingRegsiter(addr, unitID) { \n\taddr += node.splitAddress\n\tif (unitID === node.unitId && \n\t\taddr >= node.splitAddress && \n\t\taddr <= node.splitAddress * 2) { \n\n\t\treturn node.registers.readUInt16BE(addr * node.bufferFactor)  \n\t} \n}",
        "funcSetCoil": "function setFlexCoil(addr, value, unitID) { \n\tif (unitID === node.unitId && \n\t\taddr >= node.minAddress && \n\t\taddr <= node.splitAddress) { \n\n\t\tnode.coils.writeUInt8(value, addr * node.bufferFactor)  \n\t} \n}",
        "funcSetRegister": "function setFlexRegister(addr, value, unitID) { \n\taddr += node.splitAddress\n\tif (unitID === node.unitId && \n\t\taddr >= node.splitAddress && \n\t\taddr <= node.splitAddress * 2) { \n\n\t\tnode.registers.writeUInt16BE(value, addr * node.bufferFactor)  \n\t} \n}",
        "showErrors": false,
        "x": 570.9999923706055,
        "y": 293,
        "wires": [
            [
                "501ddac6b3c7f223"
            ],
            [
                "4b2d9eb39d207dfe"
            ],
            [
                "a01a8854cd169601"
            ],
            [
                "7c7739ac5614243e"
            ],
            [
                "5a0727dd1d723da0"
            ]
        ]
    },
    {
        "id": "75fae7a995dab7c6",
        "type": "inject",
        "z": "87e693e9a43e62a9",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "30",
        "crontab": "",
        "once": true,
        "onceDelay": "45",
        "topic": "",
        "payload": "messageSequenceNumber",
        "payloadType": "flow",
        "x": 395.99999237060547,
        "y": 293,
        "wires": [
            [
                "79f47b97b1b4ffd4"
            ]
        ],
        "l": false
    },
    {
        "id": "501ddac6b3c7f223",
        "type": "debug",
        "z": "87e693e9a43e62a9",
        "name": "debug 12",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790.9999923706055,
        "y": 213,
        "wires": []
    },
    {
        "id": "4b2d9eb39d207dfe",
        "type": "debug",
        "z": "87e693e9a43e62a9",
        "name": "debug 13",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790.9999923706055,
        "y": 253,
        "wires": []
    },
    {
        "id": "a01a8854cd169601",
        "type": "debug",
        "z": "87e693e9a43e62a9",
        "name": "debug 14",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790.9999923706055,
        "y": 293,
        "wires": []
    },
    {
        "id": "7c7739ac5614243e",
        "type": "debug",
        "z": "87e693e9a43e62a9",
        "name": "debug 15",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790.9999923706055,
        "y": 333,
        "wires": []
    },
    {
        "id": "5a0727dd1d723da0",
        "type": "debug",
        "z": "87e693e9a43e62a9",
        "name": "debug 16",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790.9999923706055,
        "y": 373,
        "wires": []
    }
]
```

### What did you expect to happen?

No Error Logged

biancode, 2022-09-25:
Hi @CTGControls - I'd say the payload is undefined in that situation.

CTGControls, 2022-10-03:
I'll see if I can grab some more information about it and maybe do a little digging if I get sometime next month. I am tied up this month.

biancode, 2022-12-09:
This is a problem from the injected msg.payload see the node docs.

closed 2023-03-10 by github-actions[bot]