BiancoRoyal/node-red-contrib-modbus #336: Breaks Node Red split join nodes

OlivierVerhaegen, 2022-10-10:
### Which node-red-contrib-modbus version are you using?

5.23.2

### What happened?

When updating to the newest version I noticed that this version breaks the split and join nodes by removing the `parts` property from the object.

[image]

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

Connect split node before Modbus Flex Getter node and join node after.

### What did you expect to happen?

For the split/join operations to work like in version 5.21.2

biancode, 2022-10-17:
Hi @OlivierVerhaegen, a flow would help to test. You can use the node-red-contrib-helper package to build a test fro our e2e testings. Please, insert comment what is expected inside the flow! Thank you in advance!

OlivierVerhaegen, 2023-02-19:
Hi @biancode , I'm sorry I haven't been able to create an example flow lately because of busy times. I'll post it here when I get to it! 

My debugging tests showed that it might be a straight forward issue. The `parts` property is normally added to an object going through a split node (to be able to reconstruct the array after looping through it). When placing a Modbus flex getter node in the chain the `parts` property is removed from the object. This causes problems with reconstructing the array in the join node.

unborn-andy, 2023-02-19:
Have you tried checking the Option "Keep Msg Properties" in the Modbus flex getter node ?

[image]

OlivierVerhaegen, 2023-04-07:
Hi Andy,

I may have been overlooking this checkmark.
Will try, thank you!

closed 2023-06-08 by OlivierVerhaegen