BiancoRoyal/node-red-contrib-modbus #337: Unable to update - Python3 missing

vantech14, 2022-10-19:
### Which node-red-contrib-modbus version are you using?

Home Assistant Node Red add-on

### What happened?

Tried to update to 5.23.2 and received the below error. 

2022-10-06T18:14:50.215Z [err] ERR! gyp ERR! find Python - "python3" is not in PATH or produced an error
2022-10-06T18:14:50.215Z [err] npm ERR! gyp ERR! find Python checking if "python" can be used
2022-10-06T18:14:50.215Z [err] npm ERR! gyp ERR! find Python - "python" is not in PATH or produced an error
2022-10-06T18:14:50.215Z [err] npm ERR! gyp ERR! find Python
2022-10-06T18:14:50.215Z [err] npm ERR! gyp ERR! find Python **********************************************************
2022-10-06T18:14:50.215Z [err] npm ERR! gyp ERR! find Python You need to install the latest version of Python.

### Server

Modbus-Server Node

### How can this be reproduced?

Trying to upgrade

biancode, 2022-12-09:
out of scope for us

closed 2022-12-09 by biancode

MxD-js, 2024-09-05:
I'm trying to install this node on a PLC and it does not have python which is needed by this package for serial communication, all I need is modbus TCP. Is it possible to remove the dependency on serial and have a tcp node separately?