BiancoRoyal/node-red-contrib-modbus #340: Writing negate integer values not supported (RangeError)

ricardoschluter, 2022-10-31:
### Which node-red-contrib-modbus version are you using?

5.23.2

### What happened?

Using FC6 or FC16 to write integer values with a "-" results in the following error error 
[image]

### Server

Other/External server

### How can this be reproduced?

Write a negative integer value to a Modbus-Write or Modbus-Flex-Write node

### What did you expect to happen?

Negative value to be written and/or to allow for defining int/uint values.

### Other Information

As per Sunspec standard, writing negative values is allowed for int16BE,in32BE (and maybe more) types. 
Testing writing the neagtive values with QModMaster on the same registers, is working.
[image]

JoelKrec, 2022-12-08:
Hello @KillerTic and @ricardoschluter ,
Your problems arise because of signed and unsigned integers.

Our package implements modbus protocoll in Node-RED with the npm package modbus-serial, which implements the official modbus specification. 
You can read that on modbus.org
There, you can see in the section 4.3 that the data of holding registers are 16 bit words. The "real" type of that data is entirely up to the application. 

Therefore, you always have to convert your data type to an unsigned integer before writing and back to your type after reading. 

We want to leave that freedom to our users.
That is why we do not want to alter/interpret any data.

In your case you interpret the data as a signed integer (from -32768 to 32767). To use our package, you need to convert that to an unsigned integer (from 0 to 65535). You can read more about that conversion on the internet.

I hope this will help the both of you and if you want more support, you can think about becoming a github sponsor or a PLUS4Node-RED member as you could get top support from us and we could bring this package forward faster and better.

biancode, 2022-12-12:
The Modbus IO-File is not well documented at the moment, but this is exactly the option to handle the data types from an IO input description and this provides the correct values with human readable names if the IO-File is well configured.

closed 2023-03-15 by github-actions[bot]