BiancoRoyal/node-red-contrib-modbus #342: serial RTU reconnecting

kgn-dkau2mation, 2022-11-15:
### Which node-red-contrib-modbus version are you using?

5.23.2

### What happened?

I'm trying to read registers from serial RTU, but the Modbus Flex Getter keeps saying "initialize" and "reconnecting after 2000 msec".
The server is set up to use "Serial Expert", and baud rates which I have confirmed using
```
mbpoll -l 100 -b 9600 -t 3 -c 9 -P none /dev/ttymxc0  -a 7 -m rtu -d 8
```

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

Import the following flow. The modbus node switches between "Initialize" and "reconnecting after 2000 msec."
Triggering timestamp gives error "Error: Client Not Ready To Read At State init".
LED on serial does not blink.

```
[
    {
        "id": "944b9da178c42860",
        "type": "debug",
        "z": "e3af0eae7da2143f",
        "name": "debug 71",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 760,
        "y": 1780,
        "wires": []
    },
    {
        "id": "adaf6eac9c0d6b1f",
        "type": "inject",
        "z": "e3af0eae7da2143f",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 160,
        "y": 1780,
        "wires": [
            [
                "d8673e507557114d"
            ]
        ]
    },
    {
        "id": "d8673e507557114d",
        "type": "function",
        "z": "e3af0eae7da2143f",
        "name": "function 21",
        "func": "var dataRequest = {};\ndataRequest.payload = {\n    value: 32,\n    fc: 3,\n    unitid: 7,\n    address: 1,\n    quantity: 8\n}\nreturn dataRequest;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 330,
        "y": 1780,
        "wires": [
            [
                "2b902de8c13de60a"
            ]
        ]
    },
    {
        "id": "2b902de8c13de60a",
        "type": "modbus-flex-getter",
        "z": "e3af0eae7da2143f",
        "name": "",
        "showStatusActivities": true,
        "showErrors": true,
        "logIOActivities": false,
        "server": "5c9bc07df74b36d0",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": true,
        "keepMsgProperties": true,
        "x": 540,
        "y": 1780,
        "wires": [
            [
                "944b9da178c42860"
            ],
            []
        ]
    },
    {
        "id": "5c9bc07df74b36d0",
        "type": "modbus-client",
        "name": "",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttymxc0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": 7,
        "commandDelay": 1,
        "clientTimeout": 1000,
        "reconnectOnTimeout": true,
        "reconnectTimeout": 2000,
        "parallelUnitIdsAllowed": false
    }
]
```

### What did you expect to happen?

State should show "Connected" or equivalent. Debug output should show the reported value. Serial LED should blink.

### Other Information

I'm using Node-RED version: v3.0.2
and Node.js version: v16.18.1

kgn-dkau2mation, 2022-11-16:
In addition to the above: When I run node-red in debug-mode (`DEBUG=contribModbus*,modbus-serial node-red -v`), I get the following (looping every 2 seconds):

```
16 Nov 09:13:15 - [warn] [modbus-client:5c9bc07df74b36d0] Client -> fsm init state after reconnecting  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7
16 Nov 09:13:15 - [warn] [modbus-client:5c9bc07df74b36d0] Client -> fsm init in 2000 ms  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7
  contribModbus:core:client Client -> connection closed  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7 +4s
  contribModbus:core:client Client -> connection close sent  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7 +0ms
  contribModbus:core:client Client -> RTU buffered port serial  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7 +1ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 5c9bc07df74b36d0 +4s
  modbus-serial modbusSerialErrorHandling:Module did not self-register: '/home/unipi/.node-red/node_modules/@serialport/bindings/build/Release/bindings.node'. +4s
16 Nov 09:13:17 - [warn] [modbus-client:5c9bc07df74b36d0] Client -> fsm broken state after init Get More About It By Logging  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7
16 Nov 09:13:17 - [warn] [modbus-client:5c9bc07df74b36d0] Client -> fsm reconnect state after broken Get More About It By Logging  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7
  contribModbus:queue:core queue serial lock command node name: undefined id: 5c9bc07df74b36d0 +3ms
```

biancode, 2022-11-16:
Hi,

the problem looks like a serialport problem see this:

Client -> RTU buffered port serial  Serial@/dev/ttymxc0:9600bit/s default Unit-Id: 7 +1ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 5c9bc07df74b36d0 +4s
  modbus-serial modbusSerialErrorHandling:Module did not self-register: '/home/unipi/.node-red/node_modules/@serialport/bindings/build/Release/bindings.node'. +4s

more than 4 sec to wait for bindings - is it an installation problem - switch of bide version without reinstalling serialport?

kgn-dkau2mation, 2022-11-16:
It was indeed an installation problem. I did the following to resolve the issue:
```
cd ~/.node-red
npm uninstall node-red-contrib-modbus
npm install node-red-contrib-modbus
```

Thanks for your help, much appreciated!

closed 2022-11-16 by kgn-dkau2mation