BiancoRoyal/node-red-contrib-modbus #347: Node does not send an error message

DmitrySidorow, 2022-12-08:
### Which node-red-contrib-modbus version are you using?

5.24.0

### What happened?

The function of sending an empty error message no longer works.
[image]

### Server

Modbus-Server Node

### How can this be reproduced?

Use modbus flex write to unreachable tcp server

### What did you expect to happen?

Node should send empty message if connection error. It helped me to keep track of communication with devices

biancode, 2022-12-08:
Hi @DmitrySidorow yes, the node has a new check for inactive connection. We can add to send out an empty message on each fail for input, but maybe it is in your case not an error from a failed Modbus function anymore.

biancode, 2022-12-08:
check the new warnings of the nodes

DmitrySidorow, 2022-12-08:
> Hi @DmitrySidorow yes, the node has a new check for inactive connection. We can add to send out an empty message on each fail for input, but maybe it is in your case not an error from a failed Modbus function anymore.

Yes, it is definitely necessary for the node to send some kind of error message so that this message can be processed. Thank you

biancode, 2022-12-09:
If you are in trouble without an empty msg, then you send to fast to the node and this is not to handle by the node.

biancode, 2022-12-12:
see #348 - it is a different meaning if the Modbus client could not send or fails or if a message is too early or in the wrong moment passed into to the node. To handle both situations it makes sense to get a new node to check if a used client is ready to accept msg. That is easier to handle in programming and the programmer knows exactly what to case is at the problem of a not to send msg. The empty msg will just be sent if the Modbus client has a problem or fails.

biancode, 2022-12-12:
In the most support requests we analyzed now, that more than 80% of the problems are too early sends into the node, while Modbus is not ready to accept and send the msg objects. It is not clear, which impact the deny of msg objects since v5.24 has in a project that uses the queue intensive. It would be great to learn more from the community about the needs in the different cases of the client and the nodes to handle the Modbus msg objects well.

unborn-andy, 2024-04-13:
Hello @biancode .. i will agree with @DmitrySidorow that its important to send an "Empty msg on fail" when the option is checked because we also have the need to process this msg in order to update the status of a Disconnected device.
The Catch node doesnt seem to catch any errors since its linked to the Modbus Read, Write, or Sequencer nodes and not the actual Modbus Client config node.

The same way you log an error in the Debug window when the option is checked to log failures, can we also have a msg when there is a Failure ?

[image]

Maybe the option "Empty msg on fail" should have been on the Modbus Client config node instead ?

I believe this feature was working before but now when the latest versions, when we have the option to "Reconnect on timeout" it seems broken since the Modbus Client doesnt accept msgs when its in Reconnect state ?

If you have time im sharing part of our flow to better visualize what we are trying to achieve

```
[{"id":"93a04dfd46cf13bc","type":"modbus-flex-sequencer","z":"89e2edc2.2ff858","name":"Modbus-Flex-Sequencer","sequences":[{"name":"","unitid":"31","fc":"FC1","address":"1","quantity":"1"}],"server":"c6d958ad.ea6e1","showStatusActivities":true,"showErrors":true,"showWarnings":true,"logIOActivities":false,"useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":true,"keepMsgProperties":true,"delayOnStart":false,"startDelayTime":"","x":650,"y":260,"wires":[["a6573f24c79b6828"],[]]},{"id":"8e27e7f4d7202131","type":"function","z":"89e2edc2.2ff858","name":"sequences","func":"let mbConfig = global.get(\"mbConfig\")\n\nif (mbConfig) {\n\n    msg.sequences = [\n        { topic: \"ELNET31a\", fc: 4, unitid: 1, address: 1, quantity: 64, specification: mbConfig.ELNET31a },\n        { topic: \"ELNET31b\", fc: 4, unitid: 1, address: 105, quantity: 12, specification: mbConfig.ELNET31b },\n        { topic: \"ELNET31c\", fc: 4, unitid: 1, address: 6799, quantity: 12, specification: mbConfig.ELNET31c },\n        { topic: \"ELNET32a\", fc: 4, unitid: 2, address: 1, quantity: 64, specification: mbConfig.ELNET32a },\n        { topic: \"ELNET32b\", fc: 4, unitid: 2, address: 105, quantity: 12, specification: mbConfig.ELNET32b },\n        { topic: \"ELNET32c\", fc: 4, unitid: 2, address: 6799, quantity: 12, specification: mbConfig.ELNET32c },\n        { topic: \"ELNET33a\", fc: 4, unitid: 3, address: 1, quantity: 64, specification: mbConfig.ELNET33a },\n        { topic: \"ELNET33b\", fc: 4, unitid: 3, address: 105, quantity: 12, specification: mbConfig.ELNET33b },\n        { topic: \"ELNET33c\", fc: 4, unitid: 3, address: 6799, quantity: 12, specification: mbConfig.ELNET33c },\n        { topic: \"ELNET34a\", fc: 4, unitid: 4, address: 1, quantity: 64, specification: mbConfig.ELNET34a },\n        { topic: \"ELNET34b\", fc: 4, unitid: 4, address: 105, quantity: 12, specification: mbConfig.ELNET34b },\n        { topic: \"ELNET34c\", fc: 4, unitid: 4, address: 6799, quantity: 12, specification: mbConfig.ELNET34c },\n        { topic: \"ELNET35a\", fc: 4, unitid: 5, address: 1, quantity: 64, specification: mbConfig.ELNET35a },\n        { topic: \"ELNET35b\", fc: 4, unitid: 5, address: 105, quantity: 12, specification: mbConfig.ELNET35b },\n        { topic: \"ELNET35c\", fc: 4, unitid: 5, address: 6799, quantity: 12, specification: mbConfig.ELNET35c },\n        { topic: \"ELNET36a\", fc: 4, unitid: 6, address: 1, quantity: 64, specification: mbConfig.ELNET36a },\n        { topic: \"ELNET36b\", fc: 4, unitid: 6, address: 105, quantity: 12, specification: mbConfig.ELNET36b },\n        { topic: \"ELNET36c\", fc: 4, unitid: 6, address: 6799, quantity: 12, specification: mbConfig.ELNET36c },\n        { topic: \"ELNET37a\", fc: 4, unitid: 7, address: 1, quantity: 64, specification: mbConfig.ELNET37a },\n        { topic: \"ELNET37b\", fc: 4, unitid: 7, address: 105, quantity: 12, specification: mbConfig.ELNET37b },\n        { topic: \"ELNET37c\", fc: 4, unitid: 7, address: 6799, quantity: 12, specification: mbConfig.ELNET37c },\n        // { topic: \"ELNET38a\", fc: 4, unitid: 38, address: 1, quantity: 64, specification: mbConfig.ELNET38a },\n        // { topic: \"ELNET38b\", fc: 4, unitid: 38, address: 105, quantity: 12, specification: mbConfig.ELNET38b },\n        // { topic: \"ELNET38c\", fc: 4, unitid: 38, address: 6799, quantity: 12, specification: mbConfig.ELNET38c },\n    ]\n\n    return msg;\n}\n\nelse {\n    node.warn(\"No modbus configuration\")\n}","outputs":1,"timeout":"","noerr":0,"initialize":"","finalize":"","libs":[],"x":370,"y":260,"wires":[["93a04dfd46cf13bc"]]},{"id":"82e29ff2de3af69f","type":"buffer-parser","z":"89e2edc2.2ff858","name":"","data":"responseBuffer.buffer","dataType":"msg","specification":"specification","specificationType":"msg","items":[{"type":"floatbe","name":"V_31_L1","offset":0,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"V_31_L2","offset":4,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"V_31_L3","offset":8,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"A_31_L1","offset":24,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"A_31_L2","offset":28,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"A_31_L3","offset":32,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"P_31_L1","offset":36,"length":1,"offsetbit":0,"scale":"1","mask":""},{"type":"floatbe","name":"P_31_L2","offset":40,"length":1," [... cut ...] reLogEnabled":false,"tcpHost":"192.168.0.41","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","serialAsciiResponseStartDelimiter":"0x3A","unit_id":"41","commandDelay":"100","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false,"showErrors":false,"showWarnings":false,"showLogs":false}]
```

unborn-andy, 2024-04-18:
We reverted to version **5.23.3** and now we get a nice tidy error msg 

[image]
 
and with a Function node we can check if there is an error (`msg.hasOwnProperty('error')`)  and update our Dashboard accordingly if a Device is offline.

I hope "Empty msg on fail" gets fixed in the latest versions also like it was in 5.23.3

closed 2024-09-18 by github-actions[bot]