BiancoRoyal/node-red-contrib-modbus #349: Modbus-flex-getter sends error messages to debug whatever setted up in config

anykeymeister, 2022-12-13:
### Which node-red-contrib-modbus version are you using?

5.25.0

### What happened?

Hello!
I cannot find a way completely disable warning and error messages to debug window.

[image]
[image]

### Server

Modbus-Flex-Server Node (Please attach Script to the next section)

### How can this be reproduced?

Just perform any request to offline or non-existed slave. It send error message "Modbus Failure On State sending Get More About It By Logging" to debug window.

[{"id":"a1be2b8d9d309176","type":"modbus-flex-getter","z":"e43e9521261189a3","name":"icva-1-5266","showStatusActivities":true,"showErrors":false,"showWarnings":false,"logIOActivities":false,"server":"cbeebea8.01fdf","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"keepMsgProperties":true,"delayOnStart":false,"startDelayTime":"","x":1210,"y":280,"wires":[[],[]],"inputLabels":["ReadRequest"]},{"id":"55ded93a447cd70b","type":"inject","z":"e43e9521261189a3","name":"Timer","props":[{"p":"payload"}],"repeat":"30","crontab":"","once":true,"onceDelay":"1","topic":"","payload":"{\"fc\":1,\"unitid\":1,\"address\":2100,\"quantity\":168}","payloadType":"json","x":1010,"y":280,"wires":[["a1be2b8d9d309176"]]},{"id":"cbeebea8.01fdf","type":"modbus-client","name":"Getter","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"5002","tcpType":"TELNET","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1000","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":false}]

### What did you expect to happen?

Nothing sent to debug window.

biancode, 2022-12-13:
Hi, that is from the FSM. We will improve that soon. This should be disabled from the client (config) node, which holds the FSM inside.

biancode, 2022-12-13:
@anykeymeister did you try to switch off the Show Logs and Show Warnings inside the client config-node?

anykeymeister, 2022-12-13:
@biancode 
Hello! 
Yep, you may look at screenshots, and there are no ticked checkboxes near to Show Logs and Show Warnings. I tried both states, and it didn't help.

biancode, 2022-12-13:
[image]

anykeymeister, 2022-12-15:
> [image]

Finally found it in server config!!!!
It's new interface, I didn't see it early. 
And it works!!

Thank you!!!

closed 2022-12-15 by anykeymeister