BiancoRoyal/node-red-contrib-modbus #350: Support for sending raw modbus messages

brolaugh, 2022-12-14:
### Describe the requested feature

I'm currently trying to use a relay in NodeRED using this lib but it appears that the relay that I'm using isn't 100% modbus compatible. However by sending the raw bytes on the modbus TTY (in a python script) I'm able to flip the relay and get CRC validated responses.
Here's an example of the of "modbus commands" that I send to close and open relay channel one.

`0b 05 00 01 00 00` 

`0b 05 00 01 FF 00`

For me personally the alternative would be to entirely drop this lib start utilizing the exec nodes to control my relay which I'd rather not do.

### Motivation

It appears that going outside of the modbus spec as been [requested before](https://github.com/BiancoRoyal/node-red-contrib-modbus/issues/334) and I guess this would be going a bit further than the previously mentioned feature request.

biancode, 2022-12-14:
Hi @brolaugh or in your case you could give node-modbus-serial (modbus-serial npm) a try how it works with a simple script with that lib. If it works well and needs some special code, then you could give that lib as global lib into node-red via the settings file and use the function node of node-red to use your working code fragment inside node-red. Please, let me know if or how it works for you with that hint! Happy coding!

biancode, 2022-12-14:
`DEBUG=*modbus* node-red -v`  can show you what we send to the TTY and what you get back

biancode, 2022-12-14:
You could also set the NODE_DEBUG env variable to include the regex of *modbus* see [this article](https://www.alxolr.com/articles/debugging-tools-and-practices-in-node-js)

brolaugh, 2022-12-14:
Playing around a little with `node-modbus-serial` made me realize that I can just set each induvidual relay channel(coil) with separate commands and that way get around the modbus incompatible parts. Was unsuccessful in setting all the relay channels (coils) at once thought since that was incompatible with modbus.

closed 2023-03-16 by github-actions[bot]