BiancoRoyal/node-red-contrib-modbus #351: Multiples Unit ID on the same TCP Server

simonR35, 2023-01-05:
### Which node-red-contrib-modbus version are you using?

5.14.0

### What happened?

Hi,

I'm trying to get multiples Unit ID on the same Modbus Server. 
My goal is to have a unique Modbus TCP Server on the listening port 502 and to have multiple Unit ID (1 to 250).

But when i'm trying to write values on Unit ID 2, this is also changing the values on Unit ID 1.

Is someone got the same result ?

Best regards,

### Server

Modbus-Server Node

### How can this be reproduced?

[{"id":"38944507.17537a","type":"tab","label":"Elsys ERS2 Lite","disabled":false,"info":""},{"id":"cc6af8ca.953998","type":"function","z":"38944507.17537a","name":"Test décodage trame","func":"var packet = msg.payload;    //Récupération de la trame\nvar longueur = (packet.length) - 1;\nmsg.longueur = longueur;\nvar n = 0;\nwhile (n != 999999)\n{\n    if (n <= longueur)\n    {\n        byte1 = packet[n].toString(2);\n        msg.byte1 = byte1;\n        //Récupère le premier octet \"sensor type\"\n        var bin1 = \"0\";\n        if (byte1.length != 8)\n        {\n            if (byte1.length == 7){\n                bin1=\"0\"+byte1;\n            }\n            if (byte1.length == 6){\n                bin1=\"00\"+byte1;\n            }\n            if (byte1.length == 5){\n                bin1=\"000\"+byte1;\n            }\n            if (byte1.length == 4){\n                bin1=\"0000\"+byte1;\n            }\n            if (byte1.length == 3){\n                bin1=\"00000\"+byte1;\n            }\n            if (byte1.length == 2){\n                bin1=\"000000\"+byte1;\n            }\n            if (byte1.length == 1){\n                bin1=\"0000000\"+byte1;\n            }\n        }    \n        else {\n             bin1=byte1;\n        }\n        msg.bin1=bin1;\n        \n        //Traitement de l'octet \"sensor type\"\n        //On récupère les deux premiers bits NOB et on calcule le nombre de bits d'offset\n        var nob1 = 0;\n        nob1 = parseInt((bin1.slice(0,2)),2);\n        msg.nob1 = nob1;\n        switch (nob1){\n            case 0 :   //Si il n'y aucun bit d'offset\n                tempo = 0;\n                break;\n            case 1 :  //S'il y a un seul octet d'offset\n                tempo = packet[(n + 2)].toString(16);\n                tempo = parseInt(tempo,16);\n                break;\n            case 2 :  //S'il y a 2 octest d'offset\n                tempo1 = packet[(n + 2)].toString(16);\n                tempo2 = packet[(n + 3)].toString(16);\n                if (tempo2.length == 1){\n                    tempo2 = \"0\"+ tempo2;\n                }\n                tempo = tempo1.concat(tempo2);\n                tempo = parseInt(tempo,16);\n                break;\n            case 3 :  //S'il y a 3 octets d'offset\n                tempo1 = packet[(n + 2)].toString(16);\n                tempo2 = packet[(n + 3)].toString(16);\n                if (tempo2.length == 1){\n                    tempo2 = \"0\"+ tempo2;\n                }\n                tempo3 = packet[(n + 4)].toString(16);\n                if (tempo3.length == 1){\n                    tempo3 = \"0\"+ tempo3;\n                }\n                tempo = (tempo1.concat(tempo2)).concat(tempo3);\n                tempo = parseInt(tempo,16);\n                break;\n            case 4 : //S'il y a 4 octets d'offset\n                tempo1 = packet[(n + 2)].toString(16);\n                tempo2 = packet[(n + 3)].toString(16);\n                if (tempo2.length == 1){\n                    tempo2 = \"0\"+ tempo2;\n                }\n                tempo3 = packet[(n + 4)].toString(16);\n                if (tempo3.length == 1){\n                    tempo3 = \"0\"+ tempo3;\n                }\n                tempo4 = packet[(n + 5)].toString(16);\n                if (tempo4.length == 1){\n                    tempo4 = \"0\"+ tempo4;\n                }\n                tempo = ((tempo1.concat(tempo2)).concat(tempo3)).concat(tempo4);\n                tempo = parseInt(tempo,16);\n                break;\n            default :\n                tempo = 0;\n                break;\n        }\n                        \n        //on récupère les 6 derniers bits qui correspondent au stype\n        var stype = parseInt((bin1.slice(2,8)),2);\n        msg.stype=stype;\n        //Test du type de capteur envoyé par la trame\n        switch (stype){\n            case 1 : //si le stype correspond à un capteur de température\n                byte2 = packet[(n + 1)].toString(16);\n                byte3 = packet[(n + 2)].toString(16);\n                if(byte3.length == 1){\n                    byte3 = \"0\" + byte3;\n                }\n                temp = byte2.concat(byte3);\n                msg.temp = parseInt(temp,16);\n                msg.tempotemp = tempo;\n                n += 3 + nob1;\n                break;\n            case 2 : //si le stypes correspond à un capteur d'hygrométrie\n                byte2 = packet[(n + 1)].toString(16);\n                msg.hygro = parseInt(byte2,16);\n                msg.tempohygro = tempo;\n                n += 2 + nob1;\n                break;\n            case 7 : // si le stype correspond à la tension de la batterie\n                byte2 = packet[(n + 1)].toString(16);\n                byte3 = packet[(n + 2)].toString(16);\n                if(byte3.length == 1){\n                    byte3 = \"0\" + byte3;\n                }\n                batt = byte2.concat(byte3);\n                msg.batt = parseInt(batt,16);\n                msg.tempobatt = tempo;\n                n += 3 + nob1;\n                break;\n            default : // si le stype ne correspond à aucune type configuré\n                n = 999999;\n                break;\n        }\n    }\n    else{\n        n = 999999;\n    }\n}\nreturn msg;","outputs":"1","noerr":0,"initialize":"","finalize":"","x":540,"y":200,"wires":[["c2849643.087188"]]},{"id":"dbf7c134.26663","type":"debug","z":"38944507.17537a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":290,"y":120,"wires":[]},{"id":"5541c1a5.3f5d1","type":"debug","z":"38944507.17537a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":230,"y":300,"wires":[]},{"id":"32a25b48.638904","type":"LoRa Input","z":"38944507.17537a","name":"","devEUI":"","extendedField":"","x":115,"y":200,"wires":[["5ae25a77.6b1ad4","5541c1a5.3f5d1"]],"l":false},{"id":"5ae25a77.6b1ad4","type":"Device Filter","z":"38944507.17537a","name":"","eui":"A81758FFFE08CBAA","x":250,"y":200,"wires":[["dbf7c134.26663","2784d4a0.623e5c","8e52e289.300db"]]},{"id":"2784d4a0.623e5c","type":"function","z":"38944507.17537a","name":"Payload to hex","func":"var buf = Buffer.from(msg.payload, \"base64\");\nmsg.payload = buf;\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":375,"y":200,"wires":[["cc6af8ca.953998","e1cde139.74d4f"]],"l":false},{"id":"e1cde139.74d4f","type":"debug","z":"38944507.17537a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":510,"y":120,"wires":[]},{"id":"8e52e289.300db","type":"function","z":"38944507.17537a","name":"Date & heure de la trame","func":"var time = msg.time;\nvar year = time.substring(2,4);\nvar month = time.substring(5,7);\nvar day = time.substring(8,10);\nvar hour = time.substring(11,13);\nvar minutes = time.substring(14,16);\nvar seconds = time.substring(17,19);\nmsg.ym = parseInt(year + month);\nmsg.dh = parseInt(day + hour);\nmsg.ms = parseInt(minutes + seconds);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":550,"y":260,"wires":[["3b4321a1.80920e"]]},{"id":"3d20214d.7ad45e","type":"modbus-write","z":"38944507.17537a","name":"Ecriture mesures","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"MHoldingRegisters","adr":"0","quantity":"6","server":"3aa8c4f2.a5986c","emptyMsgOnFail":false,"keepMsgProperties":false,"x":910,"y":200,"wires":[[],[]]},{"id":"c2849643.087188","type":"function","z":"38944507.17537a","name":"Mesures","func":"msg.payload = [\n    msg.temp,\n    msg.tempotemp,\n    msg.hygro,\n    msg.tempohygro,\n    msg.batt,\n    msg.tempobatt\n    ];\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":755,"y":200,"wires":[["3d20214d.7ad45e"]],"l":false},{"i [... cut ...] "type":"modbus-client","name":"Address 1","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":true}]

biancode, 2023-01-18:
Hi @simonR35, please give the Modbus Flex nodes a try to control all parameters to read or write from flexible inputs.

Zeddy-, 2023-01-31:
@biancode 

I can confirm this, just grab the example flow Multiple-Dynamic-FunctionCodes.json 

At first it seems to have different values in each unit id.. but that is just because it is updating so fast, slow the injects down to 5 second intervals and you see you in fact have exactly the same values for each unitid....

closed 2023-05-03 by github-actions[bot]

8eholder, 2023-12-20:
Hi @simonR35, I am trying to do the same but I have not had any results, I cannot emulate different slaves on a single server and port and in this or other similar cases I have not been able to see if it is viable or not. Could you solve it?

@biancode, I have tried Unit-ID 255 on the Flex Server which seems to support all Unit-IDs but does not maintain it in the holding registers. Any ideas?