BiancoRoyal/node-red-contrib-modbus #354: Read & Write nodes not Initializing with Node V18

Patrick-Work, 2023-01-15:
### Which node-red-contrib-modbus version are you using?

5.23.3 (but same issue with 5.25.0)

### What happened?

The Read and write nodes failed to initialize when the Modbus server is on the same Win10 PC after updating from Node V16.16.0 to V18.13.0. External connections still work.

### Server

Other/External server

### How can this be reproduced?

The attached flow works with Node.js 16.16.0 on Win10 pro.  (had to rename to .txt)
I ran node-v18.13.0_x64.msi to install it over V16.

The read and write nodes failed to initialize to the server. Externally everything works as expected.

Then unistalled V18 and reinstalled V16 by MSI and the flow works again.

This also happens when I disable the Node-Red Modbus server and run a Python Modbus server.
And a Flex Getter behaves the same.
[IssueWithNodeV18.txt](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/10420929/IssueWithNodeV18.txt)

### What did you expect to happen?

I expected the read & write nodes to connect to the Modbus server as they did with Node V16.

biancode, 2023-01-15:
@Patrick-Work please, try another port then 502 or try to start node-red as or with administrator rights as in Linux. Well known ports get more and more protected from newer frameworks, sdks and operating systems, which means you need to be admin or privileged user in the system to use well known ports (like 1-1024). Did you check node v18 for release notes in that point?

Patrick-Work, 2023-01-15:
> [External email: Use caution with links and attachments]

I'll give that a try, but the ports are open externally. Port 502 works with V18 when an external server is read.
Also I'll try using 127.0.01 instead of the computer name for the read nodes.

Patrick Callahan
Senior Test Engineer
Global Supply Chain
Schneider Electric
D  508-549-2385
M  508-208-5413
E  ***@***.******@***.***>
70 Mechanic Street
Foxborough, MA 02035
United States

Internal

Patrick-Work, 2023-01-15:
Yes, it's a port issue. Disabling the firewall fixes this issue. And using another port also does so. Thank you,

biancode, 2023-01-15:
@Patrick-Work you're welcome!

Patrick-Work, 2023-01-16:
Could someone guide me in how to find this issue? I created a simple Node.js server and client on the same PC using jsmodbus and they can communicate internally and externally using port 502. I tried to repeat the firewall check and it is not behaving the same as last night when I was trying multiple ports on my company network. Not sure if I triggered anything.

I have to re-open this issue as it is not a firewall issue. I can reach the port externally, unless I use the server with this package. Then it does not work.

Any guidance would be great! I have tried other modules. One worked but cannot handle more than one read or write node.

Thank you in advance,

Patrick-Work, 2023-01-16:
I ran Node-Red in debug and below is what I saw with V16 vs V18. I xxx'ed out the PC info. There seems to be another difference. The server's IP has to be 0.0.0.0 for the external connections. This was not the case with V16 as seen below.
Any advice to identify the issue is greatly appreciated. Thank you,

With Node V16:
16 Jan 17:18:00 - [info] Started modified flows
  contribModbus:server Modbus Server listening on modbus://MachineName:502 +9m
  contribModbus:server Modbus Server client connection +2s
  contribModbus:server Modbus Server client to {"address":"10.xxx.xxx.xxx","family":"IPv4","port":502} from 10.yyy.yyy.yyy 55242 +1ms

With Node V18:
16 Jan 17:30:49 - [info] Starting flows
  contribModbus:queue:core queue serial lock command node name: undefined id: 66540599b3c16b87 +0ms
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +2ms
  contribModbus:queue:core queue serial lock command node name: undefined id: 33d3fce76d1519c6 +1ms
16 Jan 17:30:49 - [info] Started flows
  contribModbus:server Modbus Server listening on modbus://MachineName:502 +0ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +22s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +1ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +25s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +0ms

Also:
16 Jan 17:34:45 - [info] Started flows
  contribModbus:server Modbus Server listening on modbus://0.0.0.0:502 +23s
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +3s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +1ms
  contribModbus:server Modbus Server client connection +3s
  contribModbus:server Modbus Server client to {"address":"10.xxx.xxx.xxx","family":"IPv4","port":502} from 10.yyy.yyy.yyy 55691 +2ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +6s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +2ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +6s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +1ms

Lastly:
16 Jan 17:36:28 - [info] Started flows
  contribModbus:server Modbus Server listening on modbus://127.0.0.1:502 +58s
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +3s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +1ms
  contribModbus:queue:core queue serial unlock command node name: undefined id: 781b141420ae0a61 +6s
  contribModbus:queue:core queue serial lock command node name: undefined id: 781b141420ae0a61 +1ms

closed 2023-04-20 by github-actions[bot]