BiancoRoyal/node-red-contrib-modbus #355: Modbus-Flex-Sequencer broken in release 5.25

hilt0n, 2023-01-16:
### Which node-red-contrib-modbus version are you using?

5.25.0

### What happened?

A valid setup of flex sequencer which was working on 5.23.2 is not working anymore with the 5.25.0. Not tested with 5.24.

I don't use the graphical feature of the sequencer and override it by injecting the sequence with data. I also tried with the sequencer example of the project. Both are not working.

The result is strange as there is no error nor answer from the node. Just nothing happens. 

### Server

Modbus-Server Node

### How can this be reproduced?

Try with the sequencer example: 
[https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/v5.25.0/examples/Modbus-Sequnecer-Demo.json](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/v5.25.0/examples/Modbus-Sequnecer-Demo.json)

### What did you expect to happen?

Receive data from the node.

### Other Information

Versions:
 - node-red: v3.0.2
 - nodejs: v16.16.0

biancode, 2023-01-18:
Hi @hilt0n, the node needs a structure of the message. The node validates that structure to exist like payload and sequences inside an incoming msg Object. You should see a warning message about the problem with an incoming msg in the verbose mode of Node-RED.

hilt0n, 2023-02-21:
Hi @biancode, sorry to not answer before. 

I enabled logging up to trace mode and there isn't any output. 

The following flow is working with previous version and not anymore in 5.25:
```
[
    {
        "id": "8ecaf6441f852002",
        "type": "tab",
        "label": "Current loops",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "eaf37da8d80b4cfd",
        "type": "inject",
        "z": "8ecaf6441f852002",
        "name": "Trigger",
        "props": [],
        "repeat": "10",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "x": 180,
        "y": 180,
        "wires": [
            [
                "e8357eb28c60bf99"
            ]
        ]
    },
    {
        "id": "e8357eb28c60bf99",
        "type": "function",
        "z": "8ecaf6441f852002",
        "name": "Fetch current",
        "func": "let sequences = []\nlet now = Date.now()\n\nfor (var i = 0; i < 6; i++) {\n    sequences[i] = {\n        'timestamp': now,\n        'fc': 3,\n        'unitid': 1 + i,\n        'address': 8,\n        'quantity': 16,\n    }\n}\nmsg.sequences = sequences\nmsg.data = now\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 350,
        "y": 180,
        "wires": [
            [
                "3b4eb8b9ea8eade8"
            ]
        ]
    },
    {
        "id": "3b4eb8b9ea8eade8",
        "type": "modbus-flex-sequencer",
        "z": "8ecaf6441f852002",
        "name": "Current loops",
        "sequences": [
            {
                "name": "dummy read",
                "unitid": "1",
                "fc": "FC1",
                "address": "1",
                "quantity": "1"
            }
        ],
        "server": "e42a34e873866683",
        "showStatusActivities": true,
        "showErrors": true,
        "showWarnings": true,
        "logIOActivities": false,
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": true,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 540,
        "y": 180,
        "wires": [
            [
                "39bebffc652b017f"
            ],
            []
        ]
    },
    {
        "id": "39bebffc652b017f",
        "type": "debug",
        "z": "8ecaf6441f852002",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 710,
        "y": 140,
        "wires": []
    },
    {
        "id": "e42a34e873866683",
        "type": "modbus-client",
        "name": "MB3480_BUS1",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "172.31.31.30",
        "tcpPort": "28561",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "0",
        "commandDelay": "1",
        "clientTimeout": "1500",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": false
    }
]```

By the way, I just tested to add a dummy value for msg.payload and it works. May be you can update the documentation as there is no mention anymore about that.

Thank you for the great work !

thalesmaoa, 2023-04-09:
I was trying it today. I can't reproduce it but didn't check your code. Try to check empty msg on error. Probably your device is off.

closed 2023-07-08 by github-actions[bot]