BiancoRoyal/node-red-contrib-modbus #358: Node-RED crash (SIGSEGV) when any writable file is set as the serial device

Matthew-Kirk, 2023-01-20:
### Which node-red-contrib-modbus version are you using?

5.25.0

### What happened?

## Version of node.js? (should be >= 16)
v16.16.0

## Version of Node-RED? (should be >= 3)
v3.0.2

## Version of node-red-contrib-modbus? (should be >= 5)
v5.25.0

## What is your platform? (Linux, macOS, ...)
Linux (Balena OS on Raspberry Pi)

Using the official Docker image

## What does DEBUG=contribModbus:*,modbus-serial* node-red -v log? (log files are welcome)
```
Welcome to Node-RED
===================

20 Jan 10:54:50 - [info] Node-RED version: v3.0.2
20 Jan 10:54:50 - [info] Node.js  version: v16.16.0
20 Jan 10:54:50 - [info] Linux 5.10.95-v8 arm64 LE
20 Jan 10:54:54 - [info] Loading palette nodes
20 Jan 10:54:55 - [debug] Module: node-red-contrib-modbus 5.25.0 /data/node_modules/node-red-contrib-modbus
20 Jan 10:54:55 - [debug] Module: node-red-dashboard 3.3.1 /data/node_modules/node-red-dashboard
20 Jan 10:54:55 - [debug] Module: node-red-contrib-stoptimer-varidelay 0.5.2 /usr/src/node-red/node_modules/node-red-contrib-stoptimer-varidelay
20 Jan 10:55:00 - [info] Dashboard version 3.3.1 started at /ui
20 Jan 10:55:00 - [info] Settings file  : /data/settings.js
20 Jan 10:55:00 - [info] Context store  : 'default' [module=memory]
20 Jan 10:55:00 - [info] User directory : /data
20 Jan 10:55:00 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
20 Jan 10:55:00 - [info] Flows file     : /data/flows.json
20 Jan 10:55:00 - [info] Server now running at http://127.0.0.1:1880/
20 Jan 10:55:00 - [debug] loaded flow revision: ca1b2db29b26aac1d5cc662158cbc46c
20 Jan 10:55:00 - [debug] red/runtime/nodes/credentials.load : no user key present
20 Jan 10:55:00 - [debug] red/runtime/nodes/credentials.load : using default key
20 Jan 10:55:00 - [debug] red/runtime/nodes/credentials.load : keyType=system
20 Jan 10:55:00 - [trace] runtime event: {"id":"runtime-state","retain":true}
20 Jan 10:55:00 - [trace] runtime event: {"id":"runtime-deploy","payload":{"revision":"ca1b2db29b26aac1d5cc662158cbc46c"},"retain":true}
20 Jan 10:55:01 - [info] Starting flows
20 Jan 10:55:01 - [debug] red/nodes/flows.start : starting flow : global
20 Jan 10:55:01 - [debug] red/nodes/flows.start : starting flow : 2ce6df32-439c-428e-bc8a-bfc3053d71e7
20 Jan 10:55:01 - [debug] red/nodes/flows.start : starting flow : f2a00bec621417c5
20 Jan 10:55:01 - [trace] [flow:global] start flow [global]
  contribModbus:queue:core queue serial lock command node name: Modbus Server id: 2995c31d.ca614c +0ms
20 Jan 10:55:01 - [info] Started flows
./entrypoint.sh: line 14:     7 Segmentation fault      (core dumped) /usr/local/bin/node $NODEOPTIONS node_modules/node-red/red.js --userDir /data $FLOWS
```

(excess details like flow logs omitted for brevity as the flow is below, though they can be given if needed)

## Flow file
```json
[{"id":"13b3589542b6b0c9","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"ec1c1510c31267ba","type":"modbus-read","z":"13b3589542b6b0c9","name":"Register 35","topic":"test","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"4","dataType":"HoldingRegister","adr":"35","quantity":"1","rate":"2","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"2995c31d.ca614c","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"x":250,"y":180,"wires":[["79748a843185534c","27f6aa74102b975f"],["b4fc3298987d3c32"]]},{"id":"8702895103125274","type":"debug","z":"13b3589542b6b0c9","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":810,"y":200,"wires":[]},{"id":"b4fc3298987d3c32","type":"debug","z":"13b3589542b6b0c9","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":500,"y":240,"wires":[]},{"id":"af697262864bf4e7","type":"modbus-read","z":"13b3589542b6b0c9","name":"Register 4","topic":"test","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"unitid":"4","dataType":"HoldingRegister","adr":"4","quantity":"1","rate":"2","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"2995c31d.ca614c","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"x":320,"y":580,"wires":[["99515002f3069944","23d180926aa53f32"],["a0be5ee50b109ba3"]]},{"id":"781b43de616918a9","type":"debug","z":"13b3589542b6b0c9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":710,"y":580,"wires":[]},{"id":"a0be5ee50b109ba3","type":"debug","z":"13b3589542b6b0c9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":580,"y":640,"wires":[]},{"id":"79748a843185534c","type":"function","z":"13b3589542b6b0c9","name":"","func":"msg.payload = \"Register 35: \" + msg.payload; \nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":500,"y":180,"wires":[["8702895103125274"]]},{"id":"02b774a462fffc01","type":"modbus-write","z":"13b3589542b6b0c9","name":"Write register 35","showStatusActivities":false,"showErrors":false,"unitid":"4","dataType":"HoldingRegister","adr":"35","quantity":"1","server":"2995c31d.ca614c","emptyMsgOnFail":false,"keepMsgProperties":false,"x":600,"y":380,"wires":[["ed3daaa4a8ac7105"],[]]},{"id":"ed3daaa4a8ac7105","type":"debug","z":"13b3589542b6b0c9","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":790,"y":380,"wires":[]},{"id":"99515002f3069944","type":"function","z":"13b3589542b6b0c9","name":"","func":"msg.payload = \"Register 4: \" + msg.payload; \nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":540,"y":580,"wires":[["781b43de616918a9"]]},{"id":"0483cb81e9d3f38c","type":"ui_text_input","z":"13b3589542b6b0c9","name":"Timer","label":"Timer","tooltip":"","group":"2e9652e4.a9280e","order":1,"width":0,"height":0,"passthru":true,"mode":"text","delay":"800","topic":"topic","topicType":"msg","x":230,"y":380,"wires":[["9c220ab025ac2356"]]},{"id":"9c220ab025ac2356","type":"function","z":"13b3589542b6b0c9","name":"Parse to Int","func":"msg.payload = parseInt(msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":410,"y":380,"wires":[["02b774a462fffc01"]]},{"id":"b6de73b7886f2348","type":"ui_text","z":"13b3589542b6b0c9","group":"2e9652e4.a9280e","order":2,"width":0,"height":0,"name":"Register 35 (Timer)","label":"Register 35 (Timer)","format":"{{msg.payload}}","layout":"row-spread","x":770,"y":80,"wires":[]},{"id":"27f6aa74102b975f","type":"function","z":"13b3589542b6b0c9","name":"Parse to Int","func":"msg.payload = parseInt(msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":510,"y":80,"wires":[["b6de73b7886f2348"]]},{"id":"aa4a4429ab779ab9","type":"modbus-write","z":"13b3589542b6b0c9","name":"Write register 4","showStatusActivities":false,"showErrors":false,"unitid":"4","dataType":"HoldingRegister","adr":"4","quantity":"1","server":"2995c31d.ca614c","emptyMsgOnFail":false,"keepMsgProperties":false,"x":560,"y":1120,"wires":[["2ec14b43288c7c88"],["56007b081fd4362c","f8fbfcd73863c590"]]},{"id":"2ec14b43288c7c88","type":"debug","z":"13b3589542b6b0c9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":790,"y":1060,"wires":[]},{"id":"d4d4914fd15a96d0","type":"ui_text_input","z":"13b3589542b6b0c9","name":"Valve","label":"Valve","tooltip":"","group":"e405fe8f.3d27a","order":1,"width":0,"height":0,"passthru":true,"mode":"text","delay":"800","topic":"topic","sendOnBlur":true,"className":"","topicType":"msg","x":110,"y":1120,"wires":[["86f3485603e93e87","961b4efa4a4118b6"]]},{"id":"86f3485603e93e87","type":"function","z":"13b3589542b6b0c9","name":"Parse to Int","func":"var msg1 = {};\nmsg1.payload = parseInt(msg.payload);\nreturn msg1;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":310,"y":1120,"wires":[["aa4a4429ab779ab9","fe313863b8d149ab"]]},{"id":"6408fcc38d16960f","type":"ui_text","z":"13b3589542b6b0c9","group":"e405fe8f.3d27a","order":2,"width":0,"height":0,"name":"Register 4 (Valve)","label":"Register 4 (Valve)","format":"{{msg.payload}}","layout":"row-spread","x":750,"y":520,"wires":[]},{"id":"23d180926 [... cut ...] is writable. E.g., if the serial port is set to `/tmp/foobar` when it does not exist, Node-RED is fine, but as soon as a regular file is created with that name, Node-RED crashes in the same way.

We have been able to replicate this with v2 and v3 of Node-RED.

### Server

Modbus-Server Node

### How can this be reproduced?

* Import the flow 
* Set the serial port on the Modbus server to an appropriate device file, or any extant regular file
* Deploy the changes and watch the logs for a segfault

Matthew-Kirk, 2023-02-17:
Hi there,

To give an update to this: we found the problem actually lies within the https://www.npmjs.com/package/modbus-serial package while running on Alpine Linux. Moving from the official Node-RED Docker Hub package (which uses Alpine) to Debian fixed the issue.

My guess is something in the chain depends on a peculiarity or a non-standard feature within glibc that musl-powered Alpine gets stuck on.

This issue can probably be closed, perhaps with a note somewhere that there may be issues using this package on Alpine?

papakpmartin, 2023-03-02:
I'm seeing the same when using the official Node-RED Docker Hub package.

I agree... if this really can't work with those official Docker images, it should probably be noted somewhere since this project is specifically a Node-RED module.

papakpmartin, 2023-03-05:
Resolved this issue for me

`node-red-contrib-modbus` works fine with official Node-RED Docker images if the npm command includes `--unsafe-perm --build-from-source`.

So in my case, in my Dockerfile I changed...

`RUN cd /data && npm install --only=production`

...to...

`RUN cd /data && npm install --only=production --unsafe-perm --build-from-source`

Also, this works for both versions 5.21 and 5.25.

mathis92, 2023-03-15:
> 

Could you please post the actual dockerfile you are using that works for you ? 
thanks

papakpmartin, 2023-03-16:
Sure! Here's the basics of what is now working for me:

```dockerfile
FROM nodered/node-red:2.2.2-14

# Copy package.json to /data so npm installs the added modules for Node-RED
COPY package.json /data/package.json

# Install inside /data. Doing otherwise may wreck the Node-RED installation
RUN cd /data && npm install --only=production --unsafe-perm --build-from-source

# Copy _your_ Node-RED project files into place
COPY settings.js /data/settings.js
COPY flows.json /data/flows.json

# Start the container normally
CMD ["npm", "start"]

closed 2023-06-14 by github-actions[bot]