BiancoRoyal/node-red-contrib-modbus #368: Incorrect and iterating value from modbus server

DirkMu, 2023-02-12:
### Which node-red-contrib-modbus version are you using?

5.25.0

### What happened?

I am using 3 'Modbus - read' nodes reading three values from one server (tcp). Poll rate is 60 seconds. I see that the values read by node red sometimes differ from a modbus simulator. In addition the values iterating. That means a new value is read (e.g. 1536). 60 seconds a new value is read e.g. 1664. Next time 1536 is read again.  

This is dump from my database sorted descending (oen example). The data is from a SMA inverter. In particular at the end of a day when the value from the inverter do not change the value in node red is iterating between the last and penultimate value (as you can see at the end of the list. 

sensor.pv_gesamtertrag_wh	2816
sensor.pv_gesamtertrag_wh	2560
sensor.pv_gesamtertrag_wh	2664
sensor.pv_gesamtertrag_wh	2560
sensor.pv_gesamtertrag_wh	2304
sensor.pv_gesamtertrag_wh	2048
sensor.pv_gesamtertrag_wh	1792
sensor.pv_gesamtertrag_wh	1536
sensor.pv_gesamtertrag_wh	1664
sensor.pv_gesamtertrag_wh	1536
sensor.pv_gesamtertrag_wh	1641
sensor.pv_gesamtertrag_wh	1536
sensor.pv_gesamtertrag_wh	unknown
sensor.pv_gesamtertrag_wh	1536
sensor.pv_gesamtertrag_wh	1280
sensor.pv_gesamtertrag_wh	1440
sensor.pv_gesamtertrag_wh	1280
sensor.pv_gesamtertrag_wh	1406
sensor.pv_gesamtertrag_wh	1280
sensor.pv_gesamtertrag_wh	1389
sensor.pv_gesamtertrag_wh	1280
sensor.pv_gesamtertrag_wh	1371
sensor.pv_gesamtertrag_wh	1280
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1256
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1208
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1185
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1181
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1178
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024
sensor.pv_gesamtertrag_wh	1177
sensor.pv_gesamtertrag_wh	1024

### Server

Other/External server

### How can this be reproduced?

I don't know...

### What did you expect to happen?

Correct value read from the modbus server.

martinroehder, 2023-03-03:
Hi,
I can support this bug. Same situation reading also reading SMA inverters. Using the "Modbus Poll" application everthing is fine.

In some cases I see the exact behavior as described above.
Reading other registers I have inbetween the correct values discrete value of exactly 0, 256 or 512.

I have the same result with two different setups. One has version 5.16. The other 5.25.

Please help us to get this stable again.

Best regards!
[image]

martinroehder, 2023-03-04:
I analysed my values a bit more. The wrong values are always a multiple of 256.
This also fits with the values of @DirkMu.

martinroehder, 2023-03-06:
I debugged it further. The error already occurs in the underlying package "modbus-serial".

martinroehder, 2023-03-07:
Found it and fixed it. The error occurs if the TCP window is too small for the whole answer and so the in this case last byte is cut off. As a counter measure modbus-serial offers a buffered TCP mode.

This mode is also available in this package. In your modbus configuration select TCP TYPE: "RTU-BUFFERED"

or you simply change the following line in my flows.json at the modbus-serial configuration:
From:  "tcpType": "DEFAULT",
To:  "tcpType": "TCP-RTU-BUFFERED",

I hope this helps also somebody else.

DirkMu, 2023-03-08:
@martinroehder thanks for debugging it. I am currently running a test. So far, it looks promising.

DirkMu, 2023-03-10:
@martinroehder with RTU-BUFFERED it is working correctly. Thank you.

closed 2023-03-10 by DirkMu