BiancoRoyal/node-red-contrib-modbus #373: modbus-client configuration node, cannot disable warnings and logs

mt-mrx, 2023-03-16:
### Which node-red-contrib-modbus version are you using?

5.25.0

### What happened?

Hi,

I've got a PV Inverter system which I'm querying via modbus/TCP. That inverter is sleeping at night (completely offline).
Since the modbus-client tries to reconnect to it very frequently and I get a log message inside the node-red debug window and the logfile I wanted to disable the "Warnings" and "Logs" since there is no way to disable the modbus-client configuration node during that time period, however disabling the options doesn't work.

### Server

Other/External server

### How can this be reproduced?

When I go into the modbus-client configuration node of that inverter to the "Optionals" tab the options "Show Warnings" and "Show Logs" are active.
To disable those messages I deselect the checkboxes, click "Update" and deploy either the "Modified Nodes" or "Full". 
a) The messages don't stop coming and b) if I go back into the configuration node both options "Show Warnings" and "Show Logs" are active again.

### What did you expect to happen?

When disabling the "Show Warnings" and "Show Logs" options of the modbus-client a) the settings should be saved and b) the messages should no longer be printed to the debug window nor log.

### Other Information

Node-red is running inside ioBroker docker container. Versions are:

ioBroker: 6.3.5
node-red adapter: 4.0.0
Node.js: v16.17.0
NPM: 8.15.0

biancode, 2023-03-16:
Hi @mt-mrx you could set up a server inside node-red and switch over for a time-range based on sun timings to have a valid zero filled connection or filled up with the last state of the inverter.

mt-mrx, 2023-03-16:
Hi @biancode thanks!

That's a workaround I can live with. Would've never got the idea to use the Modbus-Flex-Connector to switch the connection settings from the actual inverter to a dummy 
Together with node-red-contrib-sunevents it should work like a charm, just implemented it and haven't tested yet a full cycle.

If anyone is interested, here is my flow: [Dawn Dusk Connection Switcher.json](https://github.com/BiancoRoyal/node-red-contrib-modbus/files/10995607/dawn_dusk_modbus_switcher.txt)

[image]

closed 2023-06-15 by github-actions[bot]