BiancoRoyal/node-red-contrib-modbus #387: Node-RED crashing when deploying modbus read serial node

steffensoma, 2023-06-19:
### Which node-red-contrib-modbus version are you using?

5.26.0

### What happened?

When I configure a modbus read node as serial and deploy. Node-RED instantly crash and goes into infinite boot-loop. Running Node-RED in docker on Wago CC100. Tried the same on two different Wago CC100s, with same result.

`[{"id":"f90f75d5b032ff16","type":"modbus-read","z":"f6f2187d.f17ca8","name":"","topic":"","showStatusActivities":false,"logIOActivities":false,"showErrors":false,"showWarnings":true,"unitid":"","dataType":"HoldingRegister","adr":"4000","quantity":"2","rate":"5","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"aa079884c2adadbd","useIOFile":false,"ioFile":"","useIOForPayload":false,"emptyMsgOnFail":false,"x":310,"y":160,"wires":[["dad3e6a31197d94b"],[]]},{"id":"aa079884c2adadbd","type":"modbus-client","name":"MID","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"queueLogEnabled":false,"failureLogEnabled":true,"tcpHost":"192.168.1.194","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttySTM1","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"even","serialConnectionDelay":"100","serialAsciiResponseStartDelimiter":"0x3A","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectOnTimeout":true,"reconnectTimeout":"2000","parallelUnitIdsAllowed":true,"showWarnings":true,"showLogs":true}]`

### Server

Modbus-Server Node

### How can this be reproduced?

Add a single modbus read node and a debug node, configure as serial, then deploy.

### What did you expect to happen?

Node-RED not crashing and be able to read the output of the modbus node.

### Other Information

Linux CC100-568AF0 5.10.21-rt34w03.01.06 #1 SMP PREEMPT_RT Wed Jun 1 08:31:39 UTC 2022 armv7l GNU/Linux
Wago CC100 Firmware Revision 03.09.07(21)
NodeJS v16.20.0
Docker version 20.10.14-wago, build f0df241
Node-RED 3.0.2

funkitup, 2023-06-23:
I can reproduce this error on my machine.

Node-Red v3.0.2
Nodejs v16.11.1
Linux 4.14.142

AlessandroAU, 2023-06-26:
This should fix it https://github.com/BiancoRoyal/node-red-contrib-modbus/pull/390

funkitup, 2023-07-12:
> This should fix it #390

Thanks Allesandro!
However, I always get a segmentation fault after updating to Serialport@11.0.0. Any idea what's going wrong?

closed 2023-10-12 by github-actions[bot]