BiancoRoyal/node-red-contrib-modbus #410: VERY NEW VERSION Q4-2023

biancode, 2023-08-19:
### Describe the requested feature

Hi everyone,

it is time to build a "VERY NEW VERSION Q4-2023" until that time we do not fix issues here, but we will watch all your issues to test and improve them in the next very new version released by PLUS for Node-RED (P4NR) in the namespace plus4nodered.

### Motivation

It is a long time now as I started to maintain this in my spare time and now as a company leader and FlowFuse partner it is time to provide Open Source solutions for Node-RED and FlowFuse and we have a team now to do it with all our power and knowledge to provide the next best version of Modbus for Node-RED in industry and IIoT use cases.

biancode, 2023-10-24:
This package is now maintained by the PLUS for Node-RED (P4NR) B2B Community.

https://plus4nodered.com/

biancode, 2024-01-27:
It takes a bit longer and our new goal is "VERY NEW VERSION Q2-2024"

dansonamission, 2024-04-03:
Will this new version exclude the use of VM2?

biancode, 2024-04-04:
> Will this new version exclude the use of VM2?

Yes, we will use the normal function node of Node-RED in that new version.

biancode, 2024-05-04:
#461 depends on this

hamuryen, 2024-05-31:
Hello,

We would like to use the `node-red-contrib-modbus` module within Node-RED, but we cannot do so due to the reported vulnerability at [this link](https://github.com/advisories/GHSA-cchq-frgv-rjh5). Checking the GitHub repository, it seems that the `vm2` dependency has been removed, but there hasn't been any release since then. Could you let us know when the next release is planned?

Thanks.

biancode, 2024-05-31:
> Hello,
>
> We would like to use the `node-red-contrib-modbus` module within Node-RED, but we cannot do so due to the reported vulnerability at [this [...]

@hamuryen we working on that and the split is done, but many tests used the Flex-Server and we have to write new tests to get back to the good code coverage. v5.40.x will be released soon

closed 2024-08-30 by github-actions[bot]