BiancoRoyal/node-red-contrib-modbus #411: send binary value

mjmokhtar, 2023-08-26:
### Describe the requested feature

i know modbus use binary 16 bit 
example I want to sent 0000000000000010

what is program 
why result in modbus become integer

### Motivation

PLC Beckhoff & Siemen alwasy use binary value to start some engine

biancode, 2023-08-26:
you need to write the data as buffer if you can use the normal function codes for it and at the end all is 16Bit buffer data and you get the raw buffer on output two to convert and read it as what ever you want to read it 1Bit - ...Bit

mjmokhtar, 2023-08-26:
i found this if i send a buffer

```
// Create a buffer to hold the 16-bit value
var buffer = Buffer.alloc(2);
buffer.writeUInt16BE(decimalValue, 0);

```

[image]

i don't know this code was right or wrong

mjmokhtar, 2023-08-26:
did node-red-contrib-modbus support fc 6 for write single register using binary 16 bit?

mjmokhtar, 2023-08-31:
my problem solve, before using modbus make sure your plc (beckhoff, allen bradly, or siemens)  was configurate for modbus tcp/ip or modbus rtu
thank you

closed 2023-08-31 by mjmokhtar