BiancoRoyal/node-red-contrib-modbus #413: no response object

toplinuxsir, 2023-09-08:
### Which node-red-contrib-modbus version are you using?

 with 5.26

### What happened?

When I convert modbus read to http endpoint , the http request timeout , I can confirm the modubs server works ok 
```
[
    {
        "id": "59ff2a1.fa600d4",
        "type": "http in",
        "z": "63399198ad969e4a",
        "name": "",
        "url": "/hello",
        "method": "get",
        "upload": false,
        "swaggerDoc": "",
        "x": 140,
        "y": 140,
        "wires": [
            [
                "966b586a54e403fc"
            ]
        ]
    },
    {
        "id": "266c286f.d993d8",
        "type": "http response",
        "z": "63399198ad969e4a",
        "name": "",
        "statusCode": "",
        "headers": {},
        "x": 610,
        "y": 120,
        "wires": []
    },
    {
        "id": "966b586a54e403fc",
        "type": "modbus-getter",
        "z": "63399198ad969e4a",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "logIOActivities": false,
        "unitid": "",
        "dataType": "Input",
        "adr": "0",
        "quantity": "8",
        "server": "6186ab15d82f8fc5",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 240,
        "y": 40,
        "wires": [
            [
                "87e9215c53d8cef3",
                "266c286f.d993d8"
            ],
            []
        ]
    },
    {
        "id": "87e9215c53d8cef3",
        "type": "modbus-response",
        "z": "63399198ad969e4a",
        "name": "",
        "registerShowMax": 20,
        "x": 490,
        "y": 220,
        "wires": []
    },
    {
        "id": "6186ab15d82f8fc5",
        "type": "modbus-client",
        "name": "art",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "10.192.2.55",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showWarnings": true,
        "showLogs": true
    }
]
```

The error prone , No response object

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

```
[
    {
        "id": "59ff2a1.fa600d4",
        "type": "http in",
        "z": "63399198ad969e4a",
        "name": "",
        "url": "/hello",
        "method": "get",
        "upload": false,
        "swaggerDoc": "",
        "x": 140,
        "y": 140,
        "wires": [
            [
                "966b586a54e403fc"
            ]
        ]
    },
    {
        "id": "266c286f.d993d8",
        "type": "http response",
        "z": "63399198ad969e4a",
        "name": "",
        "statusCode": "",
        "headers": {},
        "x": 610,
        "y": 120,
        "wires": []
    },
    {
        "id": "966b586a54e403fc",
        "type": "modbus-getter",
        "z": "63399198ad969e4a",
        "name": "",
        "showStatusActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "logIOActivities": false,
        "unitid": "",
        "dataType": "Input",
        "adr": "0",
        "quantity": "8",
        "server": "6186ab15d82f8fc5",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 240,
        "y": 40,
        "wires": [
            [
                "87e9215c53d8cef3",
                "266c286f.d993d8"
            ],
            []
        ]
    },
    {
        "id": "87e9215c53d8cef3",
        "type": "modbus-response",
        "z": "63399198ad969e4a",
        "name": "",
        "registerShowMax": 20,
        "x": 490,
        "y": 220,
        "wires": []
    },
    {
        "id": "6186ab15d82f8fc5",
        "type": "modbus-client",
        "name": "art",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "10.192.2.55",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showWarnings": true,
        "showLogs": true
    }
]
```

mjmokhtar, 2023-09-12:
i have same problem with this case

my suggestion is you can flow not in modbus output, u cant set in outside modbus for HTTP repont, or using HTTP request for send data. i hope this help you

biancode, 2023-10-23:
HTTP examples are in the menu of the Node-RED examples see Import under the main menu

closed 2024-01-22 by github-actions[bot]