BiancoRoyal/node-red-contrib-modbus #414: modbus read can inject http request without bug

mjmokhtar, 2023-09-12:
### Describe the requested feature

[image]

i have problem like that when i used modbus read like this pic

[image]

### Motivation

i  hope my request can be solved

mjmokhtar, 2023-09-16:
Antoher request modbus red can inject mysql or postgresql

mjmokhtar, 2023-09-16:
same like in top question

biancode, 2023-10-23:
Did you check the example for Modbus via HTTP in the Node-RED examples of the package if that works for you?

mjmokhtar, 2023-10-24:
i've check but dint't work its still bug

closed 2024-01-23 by github-actions[bot]