BiancoRoyal/node-red-contrib-modbus #415: Potential Bug with modbus over serial on bullseye

rxunique, 2023-09-21:
### Which node-red-contrib-modbus version are you using?

5.26.0

### What happened?

I might have found a potential bug on node red 3.1

modbus via serial tty0 works fine on boot, but it would stop working after I deploy and update to the flow

- after about 10 minutes modbus via serial would work again automatically (I tried a few times, always approx. 10 minutes)
- if I manually do node-red-stop then node-red-start, modbus would start working immediately
- this bug does not exist on buster, I tested and modbus keep working after every deploy
- I have 2 sets of hardware setup, so I was able to cross check and made sure it's not the hardware fault

When in fault, the communication to the modbus device stops completely for that 10 minute period. The whole chain is like below.

Contrib-Modbus --> Node red 3.1 --> Bullseye --> tty0 --> TTL to 485 --> modbus device

I don't know how to diagnose where it breaks down

this is also asked on node-red forum 

https://discourse.nodered.org/t/potential-bug-with-modbus-over-serial-on-bullseye/81463

### Server

Modbus-Server Node

### How can this be reproduced?

Environment
- Pi4b with bullseye 32bit
- Nodered 3.1 installed via script
- start as service on boot
- node-red-contrib-modbus 5.26.0
- bare minimum flow setup as below

`[
    {
        "id": "7c54e13bb9f3d04d",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "354c5ff55bd6b705",
        "type": "inject",
        "z": "7c54e13bb9f3d04d",
        "name": "interval",
        "props": [
            {
                "p": "topic",
                "v": "",
                "vt": "date"
            }
        ],
        "repeat": "5",
        "crontab": "",
        "once": true,
        "onceDelay": "2.5",
        "topic": "",
        "x": 260,
        "y": 500,
        "wires": [
            [
                "2342f926fad2ec1d"
            ]
        ]
    },
    {
        "id": "2342f926fad2ec1d",
        "type": "function",
        "z": "7c54e13bb9f3d04d",
        "name": "SOC",
        "func": "msg.payload = { \n    'fc': 4, \n    'unitid': 1, \n    'address': 0x311A, \n    'quantity': 1\n}\n\nreturn msg;",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 410,
        "y": 500,
        "wires": [
            [
                "be91aba91ba5f83e"
            ]
        ]
    },
    {
        "id": "09c72e188c904c36",
        "type": "modbus-response",
        "z": "7c54e13bb9f3d04d",
        "name": "View Response",
        "registerShowMax": 20,
        "x": 660,
        "y": 560,
        "wires": []
    },
    {
        "id": "be91aba91ba5f83e",
        "type": "modbus-flex-getter",
        "z": "7c54e13bb9f3d04d",
        "name": "Flex get uart",
        "showStatusActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "logIOActivities": false,
        "server": "0713c2694f47eec6",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "keepMsgProperties": false,
        "delayOnStart": false,
        "startDelayTime": "",
        "x": 550,
        "y": 500,
        "wires": [
            [
                "fb208dc46429ba1d"
            ],
            [
                "09c72e188c904c36"
            ]
        ]
    },
    {
        "id": "fb208dc46429ba1d",
        "type": "debug",
        "z": "7c54e13bb9f3d04d",
        "name": "debug 1",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 760,
        "y": 480,
        "wires": []
    },
    {
        "id": "0713c2694f47eec6",
        "type": "modbus-client",
        "name": "Pi ttyS0",
        "clienttype": "simpleser",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyS0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "115200",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showWarnings": true,
        "showLogs": true
    }
]`

sanschaussure, 2023-09-23:
I can confirm this behavior on Ubuntu. On Ubuntu 22.04 normal operation. After upgrade to 23.04 behavior like above.

**System:**
- Ubuntu 23.04
- Node-Red 3.1.0
- nodejs 20, also testet with 18
- node-red-contrib-modbus 5.26.0
- Modbus over serial (/dev/ttyUSB0) 

With enabling Reconnection in the Modbus setup no connection either after 10 Minutes

funkitup, 2023-10-06:
Might be related to [this](https://github.com/BiancoRoyal/node-red-contrib-modbus/pull/390) issue.

As far as I am aware, it is still not fixed in the latest version.

biancode, 2023-10-23:
please keep the restrictions from serialport valid see https://serialport.io/docs/11.x.x/guide-platform-support

closed 2024-01-22 by github-actions[bot]